openbmc/openbmc #2766: Witherspoon system is going to "Quiesced" state on v2r26 with GHE driver 910.1746.20180104b

kantvijay, 2018-01-08:
Started SSH Per-Connection Server (9.126.171.99:34696).
Jan 02 16:43:21 witherspoon dropbear[2578]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Jan 02 16:43:21 witherspoon dropbear[2578]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 02 16:43:21 witherspoon dropbear[2578]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 02 16:43:21 witherspoon dropbear[2578]: Child connection from ::ffff:9.126.171.99:34696
Jan 02 16:43:21 witherspoon dropbear[2578]: Exit before auth: Error reading: Connection reset by peer
Jan 02 16:43:21 witherspoon dropbear[1911]: Exit before auth: Timeout before auth
Jan 02 16:48:21 witherspoon dropbear[2584]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Jan 02 16:48:21 witherspoon dropbear[2584]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 02 16:48:21 witherspoon dropbear[2584]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 02 16:48:21 witherspoon systemd[1]: Started SSH Per-Connection Server (9.126.171.99:41019).
Jan 02 16:48:21 witherspoon dropbear[2584]: Child connection from ::ffff:9.126.171.99:41019
Jan 02 16:48:21 witherspoon dropbear[2584]: Exit before auth: Error reading: Connection reset by peer
Jan 02 16:51:07 witherspoon dropbear[1914]: Exit (root): Exited normally
Jan 02 16:51:07 witherspoon dropbear[1928]: Exit (root): Exited normally
Jan 02 16:51:07 witherspoon dropbear[1921]: Exit (root): Exited normally

Started Start Watchdog0.
Jan 05 07:12:20 witherspoon openpower-occ-control[1868]: Opening OCC device failed.
Jan 05 07:12:20 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Jan 05 07:12:21 witherspoon openpower-occ-control[1868]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::O
Jan 05 07:12:21 witherspoon openpower-occ-control[1868]:   what():  org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.
Jan 05 07:12:20 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Jan 05 07:12:20 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Jan 05 07:12:20 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Jan 05 07:12:21 witherspoon phosphor-host-state-manager[1434]: Received signal that host is running
Jan 05 07:12:21 witherspoon phosphor-host-state-manager[1434]: Change to Host State
Jan 05 07:12:21 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Jan 05 07:12:21 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Jan 05 07:12:21 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Jan 05 07:12:22 witherspoon openpower-occ-control[1876]: Opening OCC device failed.
Jan 05 07:12:22 witherspoon openpower-occ-control[1876]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::O
Jan 05 07:12:22 witherspoon openpower-occ-control[1876]:   what():  org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.
Jan 05 07:12:22 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Jan 05 07:12:22 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Jan 05 07:12:22 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Jan 05 07:12:22 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Jan 05 07:12:23 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart.
Jan 05 07:12:23 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Jan 05 07:12:23 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Jan 05 07:12:24 witherspoon openpower-occ-control[1879]: Opening OCC device failed.
Jan 05 07:12:24 witherspoon openpower-occ-control[1879]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::O
Jan 05 07:12:24 witherspoon openpower-occ-control[1879]:   what():  org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.
Jan 05 07:12:24 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Jan 05 07:12:24 witherspoon systemd[1]: Failed to start OpenPOWER OCC controller.
Jan 05 07:12:24 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Jan 05 07:12:24 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.

41:46 witherspoon phosphor-gpio-monitor[1984]: GPIO line altered
41:46 witherspoon systemd[1]: Stopped Phosphor GPIO checkstop monitor.
41:46 witherspoon systemd[1]: Stopping Phosphor poweron watchdog...
41:46 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
41:46 witherspoon systemd[1]: Starting OpenPOWER debug data collector for host c
41:47 witherspoon systemd[1]: Stopped Phosphor poweron watchdog.
41:48 witherspoon systemd[1]: Started OpenPOWER OCC Active Disable.
41:57 witherspoon checkstop_app[2484]: Host checkstop condition detected
41:57 witherspoon phosphor-log-manager[954]: Failed to find metadata
41:57 witherspoon phosphor-fru-fault-monitor[1196]: /xyz/openbmc_project/logging
41:57 witherspoon systemd[1]: Started OpenPOWER debug data collector for host ch
41:57 witherspoon systemd[1]: Reached target Quiesce Target.

kantvijay, 2018-01-08:
Logs are saved at  /esw/san2/devindia/Logs/Issue_2766  accessible from any gfw machines

geissonator, 2018-01-08:
Please open an internal CQ for this.  We'll bridge to github if it's relevant for the community.

closed 2018-01-08 by geissonator