openbmc/openbmc #2767: It takes almost 7 minutes to reboot？

lockless, 2018-01-08:
Using openbmc GUI to reboot the system，like this
[image]

then the console output as follows：
```
[  OK  ] Stopped OpenBMC Software Update Manager.
[  OK  ] Stopped Phosphor Settings Manager.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped target Network.
         Stopping Network Service...
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Phosphor Object Mapper.
         Stopping Phosphor DBus Service Discovery Manager...
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Phosphor DBus Service Discovery Manager.
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped D-Bus System Message Bus.
[ ***  ] A stop job is running for Phosphor 鈥work Manager (5min 41s / 6min 2s)
```

it use almost 7 minutes to stop the system and reboot.
Does it normal?

geissonator, 2018-01-08:
I don't think we normally see that in the reboot path, any ideas @ratagupt ?

ratagupt, 2018-01-10:
I have not seen till now that it takes too long..It would be good if you share the BMC journalctl logs..which would help in figuring it out..why it is taking too long.

gtmills, 2018-03-20:
I have not seen this. @lockless Do you have the journalctl log for this?

lockless, 2018-03-28:
For personal reasons，i have not done this research for a long time.
The log is not saved， sorry for that.
If nobody have seen this, maybe this issue can be close.

So sorry!

gtmills, 2018-08-15:
@geissonator @lockless @ratagupt Closing. If anyone sees this again, please reopen.

closed 2018-08-15 by gtmills