openbmc/openbmc #2768: Did openbmc support ipmitool?

lockless, 2018-01-08:
Today, i try to use ipmitool to access openbmc, following the steps:
```
1. download ipmitool fow windows.
2. start my openbmc system
3. config the network.( now the pc(windows system) ping opembmc is ok)
4. use the ipmitool command, but it faild
```

the output :
[image]

[image]

[image]

does the log "unable to establish LAN session" mean something?

did openbmc support ipmitool to access?
thanks!

rthomaiy, 2018-01-08:
> image

Currently using hard-coded values with no username,

Remove /-U root /from your command, and it will work.

regards,

Richard

lockless, 2018-01-08:
thanks for your answer.
I try this method, it doesn't work also.

there is the log:
[image]

rthomaiy, 2018-01-08:
I just did and it works. I am using 1.8.16, probably try verbose (-vv) 
and see the failure, Note: Authenticate NONE is should not even be 
requested, as it is not supported (Not sure, whether you are using any 
customized tool).

$ ipmitool -I lanplus -H 10.223.244.83 -P 0penBmc raw 6 1
  22 80 01 99 02 8d 57 01 00 7b 00 00 10 00 01
$ ipmitool -h
ipmitool version 1.8.16

Regards,

Richard

chiabinglee, 2018-01-09:
It seems a typo in your command, -l lanplus is wrong (not -l (little L)), it should be -I (big i) meaning "Interface".
Below is tried on my site:

[image]

lockless, 2018-01-09:
thanks for answer. that‘s so embarrassing...

I try the command with "-I" option，
there is the output.
```
root@Ubuntu:/home/skunlin# ipmitool -I lanplus -H 192.168.23.224 -P 0penBmc power status
Get Device ID command failed: 0xc1 Invalid command
Get Chassis Power Status failed: Invalid command

root@Ubuntu:/home/skunlin# ipmitool -I lanplus -H 192.168.23.224 -P 0penBmc power reset
Get Device ID command failed: 0xc1 Invalid command
Set Chassis Power Control to Reset failed: Invalid command

root@Ubuntu:/home/skunlin# ipmitool -I lanplus -H 192.168.23.224 -P 0penBmc event state
Get Device ID command failed: 0xc1 Invalid command
Get Device ID command failed: 0xc1 Invalid command
Unable to open SDR for reading
Finding sensor state... not found!
```
it always response "Invalid command"

mine260309, 2018-01-09:
I use below command and it works fine:
```
ipmitool -I lanplus -C 3 -P 0penBmc -H <bmc-ip> power status
Chassis Power is on
```

lockless, 2018-01-11:
I type the command in my device, it still can not work. always  "Invalid command"
In my bmc, process ipmid and netipmid is running:
```
root@palmetto:~# ps -a | grep ipmi
  919 root       0:00 ipmid
 1053 root       0:00 netipmid
 1060 root       0:00 grep ipmi
root@palmetto:~# 
```

network is well
```
root@Ubuntu:/home/skunlin# ping 192.168.23.224
PING 192.168.23.224 (192.168.23.224) 56(84) bytes of data.
64 bytes from 192.168.23.224: icmp_seq=1 ttl=64 time=0.414 ms
64 bytes from 192.168.23.224: icmp_seq=2 ttl=64 time=0.426 ms
64 bytes from 192.168.23.224: icmp_seq=3 ttl=64 time=0.430 ms
64 bytes from 192.168.23.224: icmp_seq=4 ttl=64 time=0.418 ms
```

then type the command it output as follows:
```
root@Ubuntu:/home/skunlin# ipmitool -vv -I lanplus -C 3 -P 0penBmc -H 192.168.23.224 power status

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04 

>> SENDING AN OPEN SESSION REQUEST

<<OPEN SESSION RESPONSE
<<  Message tag                        : 0x00
<<  RMCP+ status                       : no errors
<<  Maximum privilege level            : Unknown (0x00)
<<  Console Session ID                 : 0xa0a2a3a4
<<  BMC Session ID                     : 0x0e054053
<<  Negotiated authenticatin algorithm : hmac_sha1
<<  Negotiated integrity algorithm     : hmac_sha1_96
<<  Negotiated encryption algorithm    : aes_cbc_128

>> Console generated random number (16 bytes)
 af 73 6a 2d bb 17 ba 6c 0b c4 92 18 39 56 2b 47
>> SENDING A RAKP 1 MESSAGE

<<RAKP 2 MESSAGE
<<  Message tag                   : 0x00
<<  RMCP+ status                  : no errors
<<  Console Session ID            : 0xa0a2a3a4
<<  BMC random number             : 0xdaf920081ad699f89ae5c49d3aacc062
<<  BMC GUID                      : 0x1a67b57a20b945929e19f95ea9640d87
<<  Key exchange auth code [sha1] : 0x9b7e6afc6905436a19eb59807f76bb973f7a117f

session integrity key input (34 bytes)
 af 73 6a 2d bb 17 ba 6c 0b c4 92 18 39 56 2b 47
 da f9 20 08 1a d6 99 f8 9a e5 c4 9d 3a ac c0 62
 14 00
Generated session integrity key (20 bytes)
 90 c1 86 a6 88 2f 59 07 ad c0 f9 e9 2a a6 06 f3
 bd 20 35 fc
Generated K1 (20 bytes)
 c8 db 40 66 00 76 13 ab 66 df 81 8d 57 d3 6e 16
 97 b1 16 87
Generated K2 (20 bytes)
 52 13 d2 7c 49 c2 c4 be a9 3c 78 26 f4 b8 6e 36
 c3 93 63 c8
>> SENDING A RAKP 3 MESSAGE

<<RAKP 4 MESSAGE
<<  Message tag                   : 0x00
[... 147 lines cut ...]
No PICMG Extenstion discovered
Specified addressing     Target  0x20:0 Transit 0:0
Interface address: my_addr 0x20 transit 0:0 target 0x20:0 ipmb_target 0

RAW REQ (channel=0x0 netfn=0x6 lun=0x0 cmd=0x1 data_len=0)

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x01
>>    data    : 

Local RqAddr 0x20 transit 0:0 target 0x20:0 bridgePossible 1
**Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x1 rsp=0xc1): Invalid command**

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x3c
>>    data    : 0x6a 0x80 0xd5 0x53 

Local RqAddr 0x20 transit 0:0 target 0x20:0 bridgePossible 0
Closed Session 53d5806a

```

did i lost some config on my code or device?

gtmills, 2018-10-12:
@lockless Still a problem or can this be closed?

lockless, 2018-10-15:
This can close now.

closed 2018-10-15 by lockless