openbmc/openbmc #2769: os-release.bb.do_compile: the basehash value changed from

gtmills, 2018-01-08:
NOTE: Running task 4005 of 4012 (/home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d4ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/f88f5a8b5d438762ab76eb34ab2faa33293eb8ae/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from 72608cf46133892bd90965ab600479f6 to 41acf8081776de52d4b8a643f3f33b4f. The metadata is not deterministic and this needs to be fixed.
Build failed

sdturne, 2018-01-11:
Does anybody have any insight here?  This failure tends to occur when we run a full clean build in a new output tree, but when pointing to an existing shared-state cache from a previous build.  The failure causes the os-release file to not be generated.  Re-attempting that same build (same source) in yet another clean output tree, but against the share-state cache updated by the previous (failed) run usually succeeds (!), but the os-release file is still not generated.

gtmills, 2018-01-11:
@bradbishop Can you have a look at this when you get a chance.

sdturne, 2018-01-15:
We're now seeing that the shared state cache has nothing to do with this failure; we're seeing it on every build, and every platform we build (barreleye, romulus, witherspoon, and zaius).  We're running full clean builds with clean output trees and no pre-existing shared state cache.  Output from this morning's witherspoon build:

NOTE: recipe obmc-phosphor-image-1.ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/e5716703022a0d9ce8591a11f43f741abf265385/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from ad7e2efb5361c2980324f4371e372115 to e0825d05a629a48fca8c92390fb186f0. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/e5716703022a0d9ce8591a11f43f741abf265385/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from ad7e2efb5361c2980324f4371e372115 to e0825d05a629a48fca8c92390fb186f0. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/e5716703022a0d9ce8591a11f43f741abf265385/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from ad7e2efb5361c2980324f4371e372115 to e0825d05a629a48fca8c92390fb186f0. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/e5716703022a0d9ce8591a11f43f741abf265385/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from ad7e2efb5361c2980324f4371e372115 to e0825d05a629a48fca8c92390fb186f0. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/e5716703022a0d9ce8591a11f43f741abf265385/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from ad7e2efb5361c2980324f4371e372115 to e0825d05a629a48fca8c92390fb186f0. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/bmcbuild/releases/obmc910/Builds/extract/e5716703022a0d9ce8591a11f43f741abf265385/meta/recipes-core/os-release/os-release.bb.do_compile, the basehash value changed from ad7e2efb5361c2980324f4371e372115 to e0825d05a629a48fca8c92390fb186f0. The metadata is not deterministic and this needs to be fixed.
Build failed

gtmills, 2018-05-23:
"out-of-tree" still fails. This is still a problem.

closed 2019-04-27 by stale[bot]