openbmc/openbmc #2770: OpenBMC GUI: Codeupdate via GUI struck for more than half an hour

sivassrr, 2018-01-09:
Problem Description:
Tried to do code update via openbmc GUI, But GUI struck for more than half an hour in the same position.

[image]
[journlactlcodeupdatewsbmc015.txt](https://github.com/openbmc/openbmc/files/1615933/journlactlcodeupdatewsbmc015.txt)

sivassrr, 2018-01-10:
Tried PNOR image as well, same behaviour observed.

gtmills, 2018-02-19:
Choosing an image then pressing "Upload firmware" hangs. No commands to upload the image are sent to the BMC.

rfrandse, 2018-02-20:
<https://gerrit.openbmc-project.xyz/9056> Fix the "Upload firmware" button
Resolves  #2770 GUI: Codeupdate "Upload firmware" hangs

closed 2018-02-23 by geissonator