openbmc/openbmc #2771: OP910:DD2.1:wsbmc007:Error Log: BMC logging events at standby

cdeadmin, 2018-01-09:


spinler, 2018-01-09:
On Witherspoon, the code was creating error logs when the PS cords were pulled with power off.  That requirement was changed to only do that when the system is powered on.

rfrandse, 2018-01-24:
<https://gerrit.openbmc-project.xyz/8596> Only log input fault error when power is on
Resolves:  #2771 BMC logging power supply input fault events at standby

bjwyman, 2018-02-02:
Test results on Witherspoon system look good.

- With power off, unplug top supply.
  - Expect trace, but no error.
- Plug top power supply back in.
  - Expect clear fault trace, no error.
- Unplug bottom power supply.
  - Expect trace for bit on and fault condition, no error.
- Power on system.
  - Expect trace for bit on, fault, and create error.
- Plug in bottom power supply.
  - Expect clear fault trace, and error marked resolved.
- Unplug top power supply.
  - Expect trace for bit on, fault, and create error.
- Power off system.
- With system back in powered off state, plug in top power supply.
  - Expect trace for clear fault, and error marked resolved.

closed 2018-02-08 by geissonator