openbmc/openbmc #2772: OpenBMC GUI: Events export as CSV option missing

sivassrr, 2018-01-10:
[image]

When we were using GUI sandbox / Bluemix way, We used to observe CSV export options in Eventlog screen. Currently that is missing.

iftekharuli, 2018-03-21:
We made that export function available for sensor page. One thing to mention here though, talking to sponsor users we found out json is a much desirable format for them, not CSV.

iftekharuli, 2018-03-26:
This push https://gerrit.openbmc-project.xyz/#/c/9730/   fixes this issue

rfrandse, 2018-03-26:
<https://gerrit.openbmc-project.xyz/9730> Add Events export as json functionality
fixes  #2772 OpenBMC GUI: Events export as CSV option missing

gtmills, 2018-03-30:
The change is for exporting JSON format only. Is this okay, or is CSV format needed?  @sivassrr

sivassrr, 2018-05-30:
If we are planning to export as CSV as well, then we can have this issue/story open. Otherwise, We can close ,  I guess.

gtmills, 2018-09-19:
No plan to support CSV closing.

closed 2018-09-19 by gtmills