openbmc/openbmc #2776: Host-IPMI FRUs : SDR data structures created in memory

dkodihal, 2018-01-10:


dkodihal, 2018-01-12:
The goal of this story is to get the "IPMI FRU print" command working. Currently it doesn't because of missing SDR support. With this story, we need to see the FRU information that the host tells BMC about.

dkodihal, 2018-01-12:
We looked at the FRU locator SDR and the existing fru-gen map. Tasks based on that :

- [ ] Encode FRU entity id in the host FRU input yaml
- [ ] Script FRU entity instance
- [ ] Changes in Get SDR to look up the FRU map once it has fully read the sensors map.
- [ ] Record ids : Assign 0-255 for sensors, and 256-511 for FRUs

ratagupt, 2018-01-12:
I am updating the tasks more.

- [ ]  Encode FRU entity id and the fru entity instance in the host FRU input yaml

- [ ] Make the changes in the mako script

- [ ] Changes in Get SDR to look up the FRU map once it has fully read the sensors map.
        we need to make the logic so that each time when GET sdr command comes for a record id it should point to the next sensor/FRU.
        for the FRU we need to send the record type 11.

- [ ] Record ids : Assign 0-255 for sensors, and 256-511 for FRUs

rfrandse, 2018-01-24:
<https://gerrit.openbmc-project.xyz/8615> SDR: Adding fru records as part of GET SDR command
Resolves  #2776 Host-IPMI FRUs : SDR data structures created in memory

closed 2018-02-13 by geissonator