openbmc/openbmc #2777: Barreleye system does not build

bradbishop, 2018-01-10:
```ERROR: obmc-mgr-inventory-1.0-r1 do_unpack: Unpack failure for URL: 'file://obmc/sync_inventory_item/system/chassis/motherboard';. Unpack command PATH="/home/james/git/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin/python-native:/home/james/git/openbmc/scripts:/home/james/git/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin/allarch-openbmc-linux:/home/james/git/openbmc/build/tmp/sysroots/barreleye/usr/bin/crossscripts:/home/james/git/openbmc/build/tmp/sysroots/x86_64-linux/usr/sbin:/home/james/git/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin:/home/james/git/openbmc/build/tmp/sysroots/x86_64-linux/sbin:/home/james/git/openbmc/build/tmp/sysroots/x86_64-linux/bin:/home/james/git/openbmc/scripts:/home/james/git/openbmc/import-layers/yocto-poky/bitbake/bin:/home/james/git/openbmc-openbmc/scripts:/home/james/git/openbmc-openbmc/import-layers/yocto-poky/bitbake/bin:/home/james/.local/share/umake/bin:/home/james/bin:/home/james/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" cp -fpPRH /home/james/git/openbmc/meta-openbmc-machines/meta-openpower/meta-rackspace/meta-barreleye/recipes-phosphor/chassis/obmc-mgr-inventory%/obmc/sync_inventory_item/system/chassis/motherboard obmc/sync_inventory_item/system/chassis/ failed with return value 1
ERROR: obmc-mgr-inventory-1.0-r1 do_unpack: Function failed: base_do_unpack
ERROR: Logfile of failure stored in: /home/james/git/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-inventory/1.0-r1/temp/log.do_unpack.11375
ERROR: Task (/home/james/git/openbmc/meta-phosphor/common/recipes-phosphor/chassis/obmc-mgr-inventory.bb:do_unpack) failed with exit code '1'```

https://gerrit.openbmc-project.xyz/#/c/8283/

gtmills, 2018-07-18:
@bradbishop Barreleye has not built in months. Can we remove the layer? What is our policy for removing systems? 
https://github.com/openbmc/openbmc/tree/master/meta-openbmc-machines/meta-openpower/meta-rackspace/meta-barreleye
@KennethWilke

gtmills, 2018-07-18:
@agangidi53 ^

agangidi53, 2018-07-18:
Can we get to a middle-ground and leave that layer there ? Just in recognition of the fact that G1 was supported at some point previously ? So that people just have that history ? I do know it doesn’t build and we have a fork that you can successfully build from .

gtmills, 2018-07-18:
@agangidi53 Could you upstream that fork that does build?

gtmills, 2018-08-20:
If we remove Barreleye, we would still have the history. 
A user would just go to a previous tag or commit and could see the files.  
E.g. https://github.com/openbmc/openbmc/tree/v2.1/meta-openbmc-machines/meta-openpower/meta-rackspace/meta-barreleye

shenki, 2018-08-22:
Yes, we should remove ports that do not have an active maintainer. If we don't have someone doing builds and testing them, then it should be removed (we did the same for Firestone and Garrison a few months back).

The history will forever be part of the git repository.

@agangidi53 If you plan to submit the patches to get it building again, then let us know. Otherwise we will merge this commit in the coming week:

 https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/12072

agangidi53, 2018-08-22:
@shenki  @gtmills  Got it. Go ahead and remove. I don't think there's enough value to submitting patches to building again , given we are switching to Barreleye G2 almost entirely internally.

shenki, 2018-08-22:
Okay.

Should we talk about upstreaming Barrleye G2 so you can benefit from the latest firmware improvements on that system?

rfrandse, 2018-08-22:
<https://gerrit.openbmc-project.xyz/12072> meta-openbmc-machines: Remove Barreleye
Resolves:  #2777 Barreleye system does not build

closed 2018-08-28 by geissonator