openbmc/openbmc #2779: [Witherspoon] OCC Crash

swe12345, 2018-01-11:
Observing OCC Crash on wsbmc011.
System Info:-
```
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.109
        op-build-v1.19-316-g8d78278-dirty
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9.8
        hostboot-7050d0a
        linux-4.13.16-openpower1-pc83fb04
        petitboot-v1.6.3-pfcda8cd
        machine-xml-592a6dd
        occ-39d5490
        hostboot-binaries-6652b90
        capp-ucode-p9-dd2-v2
        sbe-b6b7bb7
```
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-164"
VERSION_ID="v2.0-164-ge4277ae"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-164"
BUILD_ID="v2.0"
```
```
| devdesc                  : Requested sensor is not present.                  |
 | moduleid                 : IPMI::MOD_IPMISENSOR                              |
 | reasoncode               : IPMI::RC_SENSOR_NOT_PRESENT                       |
 | userdata1                : BMC IPMI Completion code.                         |
 | userdata2                : bytes [0-1]sensor name bytes [2-3]sensor number by|
 |                          : tes [4-7]HUID of target.                          |   
```
```
Jan 11 14:23:19 witherspoon openpower-occ-control[1191]: Reading from OCC failed.
Jan 11 14:23:19 witherspoon openpower-occ-control[1191]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::ReadFailure'
Jan 11 14:23:19 witherspoon openpower-occ-control[1191]:   what():  org.open_power.OCC.Device.Error.ReadFailure: Reading from OCC failed.
Jan 11 14:23:19 witherspoon kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-74
Jan 11 14:23:19 witherspoon kernel[985]: [ 8685.710000] occ-hwmon occ1-dev0: occ bus op failed rc:-74
Jan 11 14:23:19 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jan 11 14:23:19 witherspoon systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Jan 11 14:23:19 witherspoon systemd[1]: Started Process Core Dump (PID 1904/UID 0).
Jan 11 14:23:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Jan 11 14:23:22 witherspoon ipmid[1244]: ERROR:[0xFF] handling NetFn:[0x4], Cmd:[0x30]
Jan 11 14:23:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Jan 11 14:23:19 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Jan 11 14:23:20 witherspoon ipmid[1244]: Error in D-Bus call
```
```
$ ./rest_cmd -i wsbmc011 -o GET -u /org/open_power/enumerate
Executing: GET  /org/open_power/enumerate

/org/open_power/enumerate
{
    "/org/open_power/control": {},
    "/org/open_power/control/gard": {},
    "/org/open_power/control/host0": {},
    "/org/open_power/control/occ0": {
        "OccActive": 1,
        "ThrottleMemTemp": 0,
        "ThrottleProcPower": 0,
        "ThrottleProcTemp": 0
    },
    "/org/open_power/control/occ1": {
        "OccActive": 0,
        "ThrottleMemTemp": 0,
        "ThrottleProcPower": 0,
        "ThrottleProcTemp": 0
    }
}
```

swe12345, 2018-01-11:
FFDC:
/gsa/ausgsa/projects/i/indiateam02/swe12345/defects/occcrash

gkeishin, 2018-01-11:
FFDC : [20180111084907041866_MyFFDCLogs.tar.gz](https://github.com/openbmc/openbmc/files/1623476/20180111084907041866_MyFFDCLogs.tar.gz)

rfrandse, 2018-01-23:
<https://gerrit.openbmc-project.xyz/8583> Retry OCC bind operations
Resolves  #2779 [Witherspoon] OCC Crashed during Power on

geissonator, 2018-02-13:
Patch out on mailing list, waiting on 4.10 merge

gkeishin, 2018-02-13:
Great. Thanks of the info @geissonator

closed 2018-03-15 by rfrandse