openbmc/openbmc #2780: openbmc build issue - openbmc/meta-phosphor/common/recipes-phosphor/mboxd/mboxd.bb:do_compile

benwei13, 2018-01-11:
I ran into this build issue with mboxd and am wondering if anyone else also ran into this

ix-map=/data/users/benwei/openbmc-ibm/openbmc/build/tmp/sysroots/romulus=  -I include -fpic -Wall -lsystemd -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o
mboxctl mboxctl-mboxctl.o
| ../git/mboxd_lpc.c: In function '__init_lpc_dev':
| ../git/mboxd_lpc.c:53:9: error: variable 'map' has initializer but incomplete type
|   struct aspeed_lpc_ctrl_mapping map = {
|          ^~~~~~~~~~~~~~~~~~~~~~~
| ../git/mboxd_lpc.c:54:3: error: unknown field 'window_type' specified in initializer
|    .window_type = ASPEED_LPC_CTRL_WINDOW_MEMORY,
|    ^
| ../git/mboxd_lpc.c:54:18: error: 'ASPEED_LPC_CTRL_WINDOW_MEMORY' undeclared (first use in this function)
|    .window_type = ASPEED_LPC_CTRL_WINDOW_MEMORY,
|                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| ../git/mboxd_lpc.c:54:18: note: each undeclared identifier is reported only once for each function it appears in
| ../git/mboxd_lpc.c:54:18: warning: excess elements in struct initializer
| ../git/mboxd_lpc.c:54:18: note: (near initialization for 'map')
| ../git/mboxd_lpc.c:55:3: error: unknown field 'window_id' specified in initializer
|    .window_id = 0, /* There's only one */
|    ^
| ../git/mboxd_lpc.c:55:16: warning: excess elements in struct initializer
|    .window_id = 0, /* There's only one */
|                 ^
| ../git/mboxd_lpc.c:55:16: note: (near initialization for 'map')
| ../git/mboxd_lpc.c:56:3: error: unknown field 'flags' specified in initializer
|    .flags = 0,
[

It appears this new structure is defined in mboxd.bb

SRC_URI += "file://aspeed-lpc-ctrl-h.patch"

but in my build environment, this patch is apparently not applied,  causing compilation failure.

I ran into this issue on both witherspoon and romulus.  If i were to remove mboxd from OpenBMC image, then my build works fine.

Any suggestions on how I may resolve this?

thanks,
-Ben

geissonator, 2018-01-11:
@cyrilbur-ibm - we saw this patch file come in via https://gerrit.openbmc-project.xyz/#/c/1774/.  Another team reported a similar issue here.  We don't see with our builds though so this is a weird one.

cyrilbur-ibm, 2018-01-15:
Bens analysis looks correct to me - yocto is not putting the `aspeed-lpc-ctrl.h` file in the correct place or not telling the compiler about it.

Who maintains yocto?

geissonator, 2018-01-15:
The yocto group does, but we def apply a lot of other patches like this fine.   We'll need a verbose bitbake of this specific recipe to see if there's something fishy going on.

cyrilbur-ibm, 2018-01-15:
I meant our yocto configuration, as you say something is going wrong in the build process.

amboar, 2018-04-18:
@benwei13 Did you resolve this issue?

fitzsim, 2018-09-06:
I worked around this issue by changing mboxd.bb to inherit autotools-brokensep instead of autotools.

closed 2019-03-27 by amboar