openbmc/openbmc #2781: witherspoon - ERROR: can't get kernel image! in QEMU

geissonator, 2018-01-11:
Loaded up the latest openbmc/qemu binary and tried to boot our latest witherspoon image.  Ran into this error:
```
-Boot 2016.07 (Dec 19 2017 - 10:49:16 +0000)

       Watchdog enabled
DRAM:  240 MiB
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=2", size 31 MiB)
ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
ubi0: min./max. I/O unit sizes: 1/1, sub-page size 1
ubi0: VID header offset: 64 (aligned 64), data offset: 128
ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 574325606
ubi0: available PEBs: 170, total reserved PEBs: 334, PEBs reserved for bad PEB handling: 0
Read 0 bytes from volume kernel-0 to 80800000
Volume kernel-0 not found!
Wrong Image Format for bootm command
ERROR: can't get kernel image!

ast# printenv
baudrate=115200
bootargs=console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=0,1 root=/dev/ubiblock0_1
bootcmd=run set_bootargs; run obmc_bootcmd
bootdelay=2
do_rwreset=if test "${rwreset}" = "true"; then ubi remove rwfs; ubi create rwfs ${rwfs_size}; fi 
ethact=aspeednic#0
kernelname=kernel-0 
loadaddr=80800000
mtddevname=u-boot
mtddevnum=0
mtdids=nor0=bmc
mtdparts=mtdparts=bmc:384k(u-boot),128k(u-boot-env),-(obmc-ubi)
obmc_bootcmd=ubi part obmc-ubi; run do_rwreset; ubi read ${loadaddr} ${kernelname}; bootm ${loadaddr} 
partition=nor0,0
root=/dev/ubiblock0_1 
rwfs_size=0x400000
set_bootargs=setenv bootargs console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=${ubiblock} root=${root} 
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
ubiblock=0,1 
verify=yes

Environment size: 926/65531 bytes
ast# ubi part obmc-ubi
ubi0: detaching mtd1
ubi0: mtd1 is detached
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=2", size 31 MiB)
ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
ubi0: min./max. I/O unit sizes: 1/1, sub-page size 1
ubi0: VID header offset: 64 (aligned 64), data offset: 128
ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 574325606
ubi0: available PEBs: 170, total reserved PEBs: 334, PEBs reserved for bad PEB handling: 0
ast# ubi info l
Volume information dump:
	vol_id          0
	reserved_pebs   33
	alignment       1
	data_pad        0
	vol_type        4
	name_len        15
	usable_leb_size 65408
	used_ebs        33
	used_bytes      2116120
	last_eb_bytes   23064
	corrupted       0
	upd_marker      0
	name            kernel-57175093
Volume information dump:
	vol_id          1
	reserved_pebs   232
	alignment       1
	data_pad        0
	vol_type        4
	name_len        13
	usable_leb_size 65408
	used_ebs        232
	used_bytes      15142912
	last_eb_bytes   33664
	corrupted       0
	upd_marker      0
	name            rofs-57175093
Volume information dump:
	vol_id          2
	reserved_pebs   65
	alignment       1
	data_pad        0
	vol_type        3
	name_len        4
	usable_leb_size 65408
	used_ebs        65
	used_bytes      4251520
	last_eb_bytes   65408
	corrupted       0
	upd_marker      0
	name            rwfs
Volume information dump:
	vol_id          2147479551
	reserved_pebs   2
	alignment       1
	data_pad        0
	vol_type        3
	name_len        13
	usable_leb_size 65408
	used_ebs        2
	used_bytes      130816
	last_eb_bytes   2
	corrupted       0
	upd_marker      0
	name            layout volume
ast# 

```

@anoo1 and @amboar took a look and it looks  like it has to do with the env variable not pointing to the appropriate name ```name            kernel-57175093```

geissonator, 2018-01-11:
FYI @saqibkh

rfrandse, 2018-01-25:
<https://gerrit.openbmc-project.xyz/8642> Fix for MACHINE_FEATURES check in u-boot.inc
Resolves  #2781 System - ERROR: can't get kernel image! in QEMU

closed 2018-01-26 by geissonator

geissonator, 2018-02-02:
Yep, I concur on that one.  It worked for the initial commit from @mtritz but future builds didn't work. He's looking into it.  Workaround is to do a "ubi info layout" and then setenv kernelname to what it shows for the kernel, then it boots like a champ.

mtritz, 2018-02-02:
I am finding that if I clone openbmc/openbmc and build a witherspoon image from scratch, the image boots in QEMU just fine - maybe the hash is cached between builds in one place, but not the other? 

We calculate it in two places - to generate UBI volumes here https://github.com/openbmc/openbmc/blob/master/meta-phosphor/classes/image_types_phosphor.bbclass#L156

and to set the environment variable here
https://github.com/openbmc/openbmc/blob/master/meta-phosphor/common/recipes-bsp/u-boot/u-boot.inc#L23

mtritz, 2018-02-02:
Per Brad, there are two issues -
1 - the get_version function is looking in STAGING_DIR_HOST, instead of STAGING_DIR_TARGET.
2 - the u-boot recipe needs to DEPEND on os-release. 

He'll be addressing these in an upcoming change.