openbmc/openbmc #2782: After the obtainment of IP address by DHCP, it would be changed to 169.254.x.x

henbin, 2018-01-12:
Hi,
As the tittle, do anyone meet the identical symptom?
Why could the link local addressing override the IP address even BMC got the valid IP address from DHCP server?

The IP address **10.99.241.174** was obtained from DHCP server.
[image]

The IP address would become **169.254.54.0** after several seconds.
[image]

The issue would disappear if I set **LinkLocalAddressing=no** in the **/etc/systemd/network/00-bmc-eth0.network**

mine260309, 2018-01-12:
OpenBMC supports [IP_aliasing](https://en.wikipedia.org/wiki/IP_aliasing), so you would see multiple IPs on an interface via `ip addr`.
`ifconfig` is deprecated, please use `ip` instead.

henbin, 2018-01-12:
Thanks @mine260309 
I got the point.
[image]

closed 2018-01-12 by henbin