openbmc/openbmc #2784: OP910:OPRAS:Witherspoon:LEDs Fail to Show Fault (Amber light) when Fan blocked with Speed 0

cdeadmin, 2018-01-12:


msbarth, 2018-01-12:
Initially produced on a witherspoon system.
In my latest recreate:

- Powered off system
- Removed blockage from fan1
- Power on system
- Fan1 enclosure = functional
- -Fan1_0 rotor = non-functional
- -Fan1_1 rotor = functional

Within the fan monitor code, at each poweron all fans(enclosures and rotors) are initialized to be functional. Now on this latest poweron, nothing was indicated that a rotor should be marked non-functional (which from the code, would mark the associated enclosure as non-functional as well which didnt happen).

At the moment, I'm leaning towards an issue within the inventory manager and its configuration to persist inventory status. This would also explain the results of the LEDs and error logging since both of these applications rely on the values stored in inventory(which currently they only look at fan enclosure functional states). Will continue to investigate down this path.

rfrandse, 2018-01-16:
<https://gerrit.openbmc-project.xyz/8491> All sensors should return a target speed value
Resolves  #2784 System:LEDs Fail to Show Fault (Amber light) when Fan blocked with Speed 0

msbarth, 2018-01-16:
@amboar is investigating the use of the max31785 TACHSEL across each revision variant to ensure the revA reports each fan rotor in a dual rotor fan assembly in the proper sysfs entries consistently which handles the case where the fan's functional state (and LEDs) are not in the correct state when the rear rotor is blocked/faulted.

msbarth, 2018-01-18:
@amboar correct, this issue is fixing a bug in regard to isolating on the front rotor, which seems to be able to be done. However, isolating on a rear rotor fault is related to what you're noting and what I've updated in #2798. I noted my initial design thoughts there in a way that we can configure-ably get around the problem within the Maxim 31785 chip and at least detect single rear rotor faults.

closed 2018-01-23 by geissonator