openbmc/openbmc #2786: [Witherspoon] Unable to GET the inventory data for the boxelder/bmc object

swe12345, 2018-01-15:
Unable to GET the inventory data for the boxelder/bmc object
```
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://wsbmc011/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
  },
  "message": "404 Not Found",
  "status": "error"
}bash-4.1$

```
System Info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-165"
VERSION_ID="v2.0-165-ge571670"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-165"
BUILD_ID="v2.0"
root@witherspoon:~#
```
```
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.109
        op-build-v1.19-316-g8d78278-dirty
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9.8
        hostboot-7050d0a
        linux-4.13.16-openpower1-pc83fb04
        petitboot-v1.6.3-pfcda8cd
        machine-xml-592a6dd
        occ-39d5490
        hostboot-binaries-6652b90
        capp-ucode-p9-dd2-v2
        sbe-b6b7bb7
root@witherspoon:~#
```

[journal_15jan.txt](https://github.com/openbmc/openbmc/files/1630677/journal_15jan.txt)

gkeishin, 2018-01-15:
@dkodihal ^^

swe12345, 2018-01-15:
```
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://xx.xx.xx.xx/xyz/openbmc_project/inventory/system/chassis/motherboard/boxeler/bmc
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
  },
  "message": "404 Not Found",
  "status": "error"
```
System Info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-164"
VERSION_ID="v2.0-164-ge4277ae"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-164"
BUILD_ID="v2.0"
root@witherspoon:~#
```

dkodihal, 2018-01-15:
Please add more details to the defect - is this seen on specific systems or all? Can you attach the eeprom content?

gkeishin, 2018-01-15:
This looks like a system specific. @swe12345

gkeishin, 2018-08-20:
No longer have data specifically for this build. Close this one and l open when it gets tested perhaps