openbmc/openbmc #2787: Immediate power off failing but GUI updates as power off

sivassrr, 2018-01-15:
[journalctlwsbmc015immpoweroff.txt](https://github.com/openbmc/openbmc/files/1631217/journalctlwsbmc015immpoweroff.txt)
On the test system, Tried immediate power off via GUI and GUI showed as off,
whereas obmcutil state as running. 
```root@witherspoon:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Running```

![immediatepoweroff](https://user-images.githubusercontent.com/19144129/34937055-72c692c4-fa09-11e7-88ab-c7c6e36a7faa.jpg)

sivassrr, 2018-01-15:
Debugged using port 2200, found host OS kernel at panic situation. 

```Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu
17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.
 .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .
.root@wsbmc015.aus.stglabs.ibm.com's password:
04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .
.  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubu
ntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.
04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .
 .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ub
untu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17
.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .
  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.0Permission denied, please try again.
root@wsbmc015.aus.stglabs.ibm.com's password:
0000b58128] panic+0x144/0x314
[  865.842961] [c000000f2918bb20] [c0000000000e21e8] do_exit+0xc68/0xc70
[  865.843022] [c000000f2918bbe0] [c0000000000e22c0] do_group_exit+0x60/0x100
[  865.843104] [c000000f2918bc20] [c0000000000f244c] get_signal+0x3bc/0x760
[  865.843170] [c000000f2918bd10] [c00000000001cea8] do_signal+0x68/0x2b0
[  865.843232] [c000000f2918be00] [c00000000001d290] do_notify_resume+0xd0/0xf0
[  865.843290] [c000000f2918be30] [c00000000000b744] ret_from_except_lite+0x70/0x74
[  865.850900] ---[ end Kernel panic - not```

sivassrr, 2018-01-15:
So opened a separate defect and send to OS screen team.

rfrandse, 2018-04-18:
<https://gerrit.openbmc-project.xyz/10204> Add fixes for power operations
Resolves  #2787 Immediate power off failing but GUI updates as power off

closed 2018-04-26 by geissonator