openbmc/openbmc #2788: Avahi spamming the journal

ratagupt, 2018-01-15:
When avahi comes up it does hostname discovery and as part of discovery it looks for unique hostname. In a network where more then 100 nodes are there,the discovery process takes a long time. as avahi uses mDNS which is incompatible with unicast DNS zones (.local).We should not use avahi in that network setup.

 More info is at the following link https://web.archive.org/web/20150228130344/http:// avahi.org/wiki/AvahiAndUnicastDotLocal

geissonator, 2018-02-13:
I think we need to get this one on the queue @dkodihal @ratagupt .  As more OpenBMC systems are shipped, the more we run into this.

rfrandse, 2018-03-20:
<https://gerrit.openbmc-project.xyz/9632> Avahi: Don't start the avahi-daemon if dot local domain is there
Resolves  #2788 Avahi spamming the journal

dkodihal, 2018-04-23:
https://gerrit.openbmc-project.xyz/#/c/9632/ has been reverted because it requires the `host` package to be present in the BMC image, and that's quite large (>2.5 MB) to have on BMCs with small flashes.

closed 2018-08-06 by bradbishop