openbmc/openbmc #2791: OCC not working on Talos systems

madscientist159, 2018-01-16:
Trying to enable core temperature monitoring on OpenBMC build for Talos systems is showing a non-functional OCC hwmon driver as follows:

On power on (obmcutil poweron)

> [  130.920000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
> [  131.760000] occ-hwmon occ2-dev0: failed to get OCC poll response: -19

On attempted manual start of OCC:

> root@talos:/sys/bus/platform/drivers/occ# occ-active.sh enable
> [  469.050000] occ-hwmon occ1-dev0: occ bus op failed rc:-74
> [  469.050000] occ-hwmon occ1-dev0: failed to get OCC poll response: -74
> [  469.050000] occ-hwmon: probe of occ1-dev0 failed with error -74

BMC is able to start the platform on its own, so the FSI bus appears to be working.  Thoughts?

amboar, 2018-01-16:
Ping @eddiejames

madscientist159, 2018-01-16:
Also seeing this on Romulus pass 2 with IBM host firmware build from late 2017.

madscientist159, 2018-01-16:
#2745 is related, but the post-IPL workaround in that report does not allow the OCC to function on our systems.

madscientist159, 2018-01-16:
One final observation: at least one of our systems cannot re-IPL after first IPL without pulling standby power from system.  Even trying to restart the FSI service yields:

> [ 2725.460000]  slave@00:00: error reading slave registers

madscientist159, 2018-01-16:
Scoping FSI data/clock lines on IPL failure after power down shows both clock and data signals toggling.  Attaching Cronus yields:

> fircheck
>  found fapi2::RC_SBE_PIB_PARTIAL_ERROR
> FAPI ERR>: /afs/rchland.ibm.com/rel/common/bringup.tools/cronus/code/dev/cronus-hg/p9/ekb_files/build_tools/../../../ecmdui/ext/fapi2/capi/hwp_ffdc_classes.H:SBE_PIB_PARTIAL_ERROR:53663: SBE SCOM failure - Partial SCOM failure
> FAPI ERR>: /afs/rchland.ibm.com/rel/common/bringup.tools/cronus/code/dev/cronus-hg/p9/ekb_files/build_tools/../../../ecmdui/ext/fapi2/capi/ffdc.H:_setHwpError:112: _setHwpError: Creating HWP error 0x81831e
>  found fapi2::RC_SBE_PIB_PARTIAL_ERROR_pcb_pib_rc
>  size: 8
>  data: 0000000000000003
> ERROR: (Cro9PuChip::isClockStateCorrect): Problems determining clock state
> ERROR: (Chip::getScom): p9n.ex:k0:n0:s0:p00:c10 domain EQ05 clocks must be on for getscom. GETSCOM on 0x1001204C
> getDoubleWord> ETRC: **** ERROR : ecmdDataBufferBase::getDoubleWord: doubleWordOffset 0 >= NumDoubleWords (0)
> FAPI ERR>: /afs/awd.austin.ibm.com/projects/cte//tools/ecmd/ver-14-8/ext/fapi2/capi/hw_access.H:getScom:116: getScom failed - Target p9n.ex:k0:n0:s0:p00:c10, Addr 000000001001204C
> FAPI ERR>: ../../chips/p9/procedures/hwp/pm//p9_query_stop_state.C:query_stop_state:197: Error reading data from CME SISR register
>  found fapi2::RC_SBE_PIB_PARTIAL_ERROR
> FAPI ERR>: /afs/rchland.ibm.com/rel/common/bringup.tools/cronus/code/dev/cronus-hg/p9/ekb_files/build_tools/../../../ecmdui/ext/fapi2/capi/hwp_ffdc_classes.H:SBE_PIB_PARTIAL_ERROR:53663: SBE SCOM failure - Partial SCOM failure
> FAPI ERR>: /afs/rchland.ibm.com/rel/common/bringup.tools/cronus/code/dev/cronus-hg/p9/ekb_files/build_tools/../../../ecmdui/ext/fapi2/capi/ffdc.H:_setHwpError:112: _setHwpError: Creating HWP error 0x81831e
>  found fapi2::RC_SBE_PIB_PARTIAL_ERROR_pcb_pib_rc
>  size: 8
>  data: 0000000000000003
> ERROR: (Cro9PuChip::isClockStateCorrect): Problems determining clock state
> ERROR: (Chip::getScom): p9n.ex:k0:n0:s0:p00:c10 domain EQ05 clocks must be on for getscom. GETSCOM on 0x1001204C
> ERROR: (Chip::getScom): p9n.ex:k0:n0:s0:p00:c11 domain EQ05 clocks must be on for getscom. GETSCOM on 0x1001204C
> getDoubleWord> ETRC: **** ERROR : ecmdDataBufferBase::getDoubleWord: doubleWordOffset 0 >= NumDoubleWords (0)
> FAPI ERR>: /afs/awd.austin.ibm.com/projects/cte//tools/ecmd/ver-14-8/ext/fapi2/capi/hw_access.H:getScom:116: getScom failed - Target p9n.ex:k0:n0:s0:p00:c11, Addr 000000001001204C
> FAPI ERR>: ../../chips/p9/procedures/hwp/pm//p9_query_stop_state.C:query_stop_state:197: Error reading data from CME SISR register
> FAPI ERR>: ../../chips/p9/procedures/utils/fir_checking//p9_fircheck.C:p9_fircheck:809: Error from walkFIRHieararchy

Is it possible for the BMC to lock up the FSI interface inside the CPUs badly enough to require a 1.1 AUX power cycle?  These are DD2.1 chips in case it matters.

geissonator, 2018-01-16:
Any thoughts @eddiejames ?

madscientist159, 2018-01-17:
Update: OCC still not working, IPL failure after chassis power off was due to an FPGA logic error and is unrelated to the OCC issue.

@eddiejames if you want to set up an interactive debug session just let me know.  We have a Cronus available if needed.

madscientist159, 2018-01-17:
This might be related.  After updating hostboot to the latest GIT master, we now see messages like this in the hostboot logs:

>  16.07304|ISTEP  8.12 - host_set_voltages
>  16.09433|================================================
>  16.09434|Error reported by sbeio (0x2800) PLID 0x9000008C
>  16.09727|  Status header does not start with magic number or non-zero primary or secondary status
>  16.09727|  ModuleId   0x02 SBEIO_FIFO
>  16.09728|  ReasonCode 0x2813 SBEIO_FIFO_RESPONSE_ERROR
>  16.09728|  UserData1  FIFO command class and command : 0x000000000000a202
>  16.09729|  UserData2  Should be magic value 0xC0DE : 0x0000c0de0005000b
>  16.09730|------------------------------------------------
>  16.09730|  Callout type             : Procedure Callout
>  16.09730|  Procedure                : EPUB_PRC_HB_CODE
>  16.09731|  Priority                 : SRCI_PRIORITY_LOW
>  16.09732|------------------------------------------------
>  16.09732|  Hostboot Build ID: 
>  16.09732|================================================

Also pdbg is unable to get CFAM data over FSI:

> pdbg -b fsi -d p9w -p0 getcfam 2809
> CRC error: 0x7f
> getcfam error. Response: 0xffffffff
> CRC error: 0x7f
> putcfam error. Response: 0xffffffff
> Couldn't open debugfs
> Couldn't open debugfs
> CRC error: 0x7f
> getcfam error. Response: 0xffffffff
> No valid targets found or specified. Try adding -p/-c/-t options to specify a target.
> Alternatively run pdbg -a probe to get a list of all valid targets

eddiejames, 2018-01-18:
If there are SBE problems then communication with the OCC won't work. The -74 error from the OCC is what I would expect if SBE wasn't working correctly. You may want to ask Sachin Gupta of SBE team?

madscientist159, 2018-01-21:
Finally sat down with this and dug into the FSI/OCC drivers.  This issue, along with the error -19 problem that occurs on FSI scan, came down to two bugs that have been fixed in our tree here:

https://git.raptorcs.com/git/talos-obmc-linux/commit/?h=dev-4.10&id=ce112cef4af78a6b47ab6542602513a9a850791c

https://git.raptorcs.com/git/talos-obmc-linux/commit/?h=dev-4.10&id=3acc96c728bcb2a9b0627466713d5d0c8b02d1d6

With these patches, we are able to read from OCC once the system has IPLed and "occ-active.sh enable" has been run.  We are working on delaying FSI scan until IPL is complete to fully resolve the remaining OCC issues.

madscientist159, 2018-01-21:
....and the OCCs are online as of [our final commit](https://git.raptorcs.com/git/talos-openbmc/commit/?id=555f9f1c1e6b724fe7de468054b14b952bbcfb58)

One thing that still puzzles me is why the SBE SCOM access returns 0x0005000b instead of the expected 0x00000000.  If this return code is valid we should mark it as an alternate valid code alongside 0x00000000.

madscientist159, 2018-01-21:
We ended up adding an OCC watchdog service here:

https://git.raptorcs.com/git/talos-openbmc/commit/?id=85ff156eca61043b1a8fb145955694bed445e678

Would like to know why skiboot knocks the OCC communication offline, but at least this works around the major problem of broken fan control.

amboar, 2018-01-21:
@madscientist159 can you please send https://git.raptorcs.com/git/talos-obmc-linux/commit/?h=dev-4.10&id=3acc96c728bcb2a9b0627466713d5d0c8b02d1d6 to the mailing list?

mine260309, 2018-03-30:
With latest OpenBMC (kernel on dev-4.13), the issue is gone.
The occ/fsi driver are updated on dev-4.13 and it seems to have fixed this issue.

amboar, 2018-04-30:
https://github.com/openbmc/openbmc/issues/2791#issuecomment-358749386 mentions hitting issues when booting the host in secure mode. The OCC driver as it stands does a putscom that's blacklisted by the SBE, so we need to do something else. http://patchwork.ozlabs.org/patch/905774/ implements the correct alternative.

mdmillerii, 2018-07-23:
@madscientist159 Is the issue resolved?  Can this be closed now?

madscientist159, 2018-07-23:
It does seem to be resolved at this point.  Thanks again!

closed 2018-07-23 by madscientist159