openbmc/openbmc #2792: IPMID: A host system event with a procedure callout

gkeishin, 2018-01-16:
Seen on every power on on Master and new fixes runs
```
/xyz/openbmc_project/logging/entry/list
[
    "/xyz/openbmc_project/logging/entry/3",
    "/xyz/openbmc_project/logging/entry/2",
    "/xyz/openbmc_project/logging/entry/1"
]
```
```
 | Reference Code           : BC8A2507                                          |
 | Hex Words 2 - 5          : 000000E0 00000300 00000000 00200000               |
 | Hex Words 6 - 9          : 00000000 000000CB CA22000B 00010000               |
 |                                                                              |
 | devdesc                  : Requested sensor is not present.                  |
 | moduleid                 : IPMI::MOD_IPMISENSOR                              |
 | reasoncode               : IPMI::RC_SENSOR_NOT_PRESENT                       |
 | userdata1                : BMC IPMI Completion code.                         |
 | userdata2                : bytes [0-1]sensor name bytes [2-3]sensor number by|
 |                          : tes [4-7]HUID of target.                          |
 |------------------------------------------------------------------------------|
```

- [x] Journald 
[journald_sensor_error.txt](https://github.com/openbmc/openbmc/files/1633738/journald_sensor_error.txt)

- [x] eSEL 
[esel_sensor.txt](https://github.com/openbmc/openbmc/files/1633743/esel_sensor.txt)

geissonator, 2018-01-16:
Looks like latest host code went in to start reading these values from the BMC.  We've got something going on in this path. Relevant commit is https://gerrit.openbmc-project.xyz/#/c/8106/.  Seems like we need to also change the /org/openbmc/sensors/host/PowerSupplyDerating path too?

dhruvibm, 2018-01-18:
This change is needed for power supply redundancy to work but that was not merged to master
https://gerrit.openbmc-project.xyz/#/c/8043/6

dkodihal, 2018-01-23:
Same as #2665.

closed 2018-01-23 by dkodihal