openbmc/openbmc #2793: Failed to start Soft power off of the host

gkeishin, 2018-01-16:
Seen during bootme run for @amboar https://gerrit.openbmc-project.xyz/#/c/8489/ fix CI run 

`Note:`  Saw failed to find metadata in the log just before the power off failure, may not be related to this but for reference, if there is co-relation in any way with this failure
```
Jan 15 20:04:44 witherspoon phosphor-log-manager[1079]: Failed to find metadata
```

` Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-166"
VERSION_ID="v2.0-166-g5df199f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.0-166"
BUILD_ID="v2.0"
root@witherspoon:~#
```
Looks like the power off hit timedout
```
{
    "data": {
        "/xyz/openbmc_project/logging/entry/10": {
            "AdditionalData": [
                "_PID=3418",
                "TIMEOUT_IN_MSEC=2700000"
            ],
            "Id": 10,
            "Message": "xyz.openbmc_project.State.Host.Error.SoftOffTimeout",
            "Resolved": 0,
            "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
            "Timestamp": 1516046684962,
            "associations": []
        }
    },
    "message": "200 OK",
    "status": "ok"
}
```

`Journald snippet`
```
Jan 15 20:04:44 witherspoon phosphor-log-manager[1079]: Failed to find metadata
Jan 15 20:04:44 witherspoon phosphor-fru-fault-monitor[1304]: /xyz/openbmc_project/logging/entry/10
Jan 15 20:04:45 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Main process exited, code=exited, status=255/n/a
Jan 15 20:04:45 witherspoon systemd[1]: Failed to start Soft power off of the host.
Jan 15 20:04:45 witherspoon systemd[1]: Dependency failed for Power0 Host Off.
Jan 15 20:04:45 witherspoon systemd[1]: obmc-host-shutdown@0.target: Job obmc-host-shutdown@0.target/start failed with result 'dependency'.
Jan 15 20:04:45 witherspoon systemd[1]: obmc-host-shutdown@0.target: Triggering OnFailure= dependencies.
Jan 15 20:04:45 witherspoon systemd[1]: usr-local.mount: Cannot add dependency job, ignoring: Cannot send after transport endpoint shutdown
Jan 15 20:04:45 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Unit entered failed state.
Jan 15 20:04:45 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Failed with result 'exit-code
```

- [x] Journald 
[journald_metadata_soft_poweroff_error.txt](https://github.com/openbmc/openbmc/files/1633760/journald_metadata_soft_poweroff_error.txt)

geissonator, 2018-01-16:
This is usually because the host hung on the power off request.  Do you have the host console output?  (i.e. https://github.com/open-power/skiboot/issues/81)

gkeishin, 2018-01-16:
I don't have one for it at the moment. I ll add one once I hit it again. Thanks

gkeishin, 2018-01-16:
Will re-open when I have more data to look at.

closed 2018-01-16 by gkeishin