openbmc/openbmc #2795: GUI: cold boot failed to restart

sivassrr, 2018-01-16:
Problem Description: 

System was at running state (OS was up and running ), Did cold boot from GUI, But system failed to power on. 

Console Text:
==========
```[   82.002671] systemd-shutdown[1]: Unmounting file systems.
[   82.469924] systemd-shutdown[1]: All filesystems unmounted.
[   82.469967] systemd-shutdown[1]: Deactivating swaps.
[   82.470039] systemd-shutdown[1]: All swaps deactivated.
[   82.470092] systemd-shutdown[1]: Detaching loop devices.
[   82.471876] systemd-shutdown[1]: All loop devices detached.
[   82.471911] systemd-shutdown[1]: Detaching DM devices.
[   82.472231] systemd-shutdown[1]: Detaching DM 253:2.
[   82.814211] shutdown: 7 output lines suppressed due to ratelimiting
[   82.854024] dracut Warning: Killing all remaining processes
dracut Warning: Killing all remaining processes
[   82.965785] XFS (dm-0): Unmounting Filesystem
[   82.974093] dracut Warning: Unmounted /oldroot.
[   82.993681] dracut: Disassembling device-mapper devices
Powering off.
[   83.081253] sd 1:0:0:0: [sdb] Synchronizing SCSI cache
[   83.081770] sd 1:0:0:0: [sdb] Stopping disk
[   83.483116] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[   83.484926] sd 0:0:0:0: [sda] Stopping disk
[   84.268074] tg3 0005:01:00.1: Refused to change power state, currently in D0
[   84.379942] reboot: Power down
[  133.284380415,5] OPAL: Shutdown request type 0x0...
[  143.284439682,3] BT: seq 0x24 netfn 0x00 cmd 0x02: Retry sending message```

```FW Details:
     ========
root@witherspoon:# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:#
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r28-0-gc9e2c09"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r28"
root@witherspoon:# cat /media/pnor-ro-7617ca43/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.113
	op-build-v1.19-320-g2364968-dirty
	buildroot-2017.08-8-g5e23247
	skiboot-v5.9.8
	hostboot-7050d0a
	linux-4.13.16-openpower1-pf94f2fe
	petitboot-v1.6.3-pc7918e8
	machine-xml-ef5f800
	occ-39d5490
	hostboot-binaries-e6e59b3
	capp-ucode-p9-dd2-v2
	sbe-b6b7bb7
root@witherspoon:#```

sivassrr, 2018-01-16:
[journalctlwsbmc004coldbootfails.txt](https://github.com/openbmc/openbmc/files/1635821/journalctlwsbmc004coldbootfails.txt)

sivassrr, 2018-01-16:
Cold boot:  Shuts down the server immediately, then restarts it - Documentation for Cold boot.

gtmills, 2018-02-07:
This is broke, working on a fix.

gtmills, 2018-02-20:
https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/State
A cold reboot should be a command to shut off the chassis 
```
$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data":"xyz.openbmc_project.State.Chassis.Transition.Off"}' https://${bmc}//xyz/openbmc_project/state/chassis0/attr/RequestedPowerTransition
```
The verify the chassis is off. (It must have time to power off, if you try to start the power on target before the off is done, problems can happen) 

Then host power on 
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.On"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
```
verify the host is on.

iftekharuli, 2018-02-21:
@gtmills  What is the command I need to run for cold reboot, orderly shutdown and immediate shutdown? I thing I have that on place but will have to make sure it is the correct one. 

Also -- can I issue this command anytime after any of the state is triggered to get the latest host update? 
```curl -c cjar -b cjar -k -X GET https://${BMC_IP}/xyz/openbmc_project/state/host0/attr/CurentHostState```

For example- I am planning to show a indeterminate progress bar when the host is being rebooted during warm reboot. So, I am assuming it should be with the same command as above I can check if the server is done rebooting and then I can update the GUI with green bar that server is running?

gtmills, 2018-02-21:
@iftekharuli Immediate shutdown calls the chassis object. https://gerrit.openbmc-project.xyz/#/c/8875/2 is the merged commit to do this. Orderly shutdown calls the host object: 
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.Off"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
```
Looking at the code, this looks correct and doesn't need a change.  
https://github.com/openbmc/phosphor-webui/blob/master/app/common/services/api-utils.js#L334

Listed above is the rest commands for cold reboot. 
It will be a call to the chassis object to power off then a call to the host object to power on. There is not one command to do a cold reboot like there is with the warm reboot. 
Yes, ```curl -c cjar -b cjar -k -X GET https://${BMC_IP}/xyz/openbmc_project/state/host0/attr/CurentHostState ```will give you the current state of the host.

iftekharuli, 2018-02-21:
Ok. What is the command for warm reboot?

gtmills, 2018-02-21:
```curl -c cjar -b cjar -k -H "Content-Type: application/json" -d '{"data": "xyz.openbmc_project.State.Host.Transition.Reboot"}' -X PUT https://${bmc}/xyz/openbmc_project/state/host0/attr/RequestedHostTransition```

https://gerrit.openbmc-project.xyz/#/c/8899/

iftekharuli, 2018-02-26:
@gtmills I couldn't make the command (curl -c cjar -b cjar -k -X GET https://${BMC_IP}/xyz/openbmc_project/state/chassis0/attr/urrentPowerState) you mentioned above work..

iftekharuli, 2018-02-26:
NV.. found the issue in the statement. Should be CurrentPowerState instead of urrentPowerState

rfrandse, 2018-02-27:
<https://gerrit.openbmc-project.xyz/9216> Add fixes for cold reboot
fixes  #2795 GUI: cold boot failed to restart

gtmills, 2018-05-15:
This should have been closed when https://gerrit.openbmc-project.xyz/#/c/9216/ was merged

closed 2018-05-15 by gtmills