openbmc/openbmc #2797: GUI: Warmboot fails to restart

sivassrr, 2018-01-16:
Problem Description:

System was at running state (OS was up and running ), Did warm boot from GUI, But system failed to restart and back to running state. 

Console Text:
```
[    4.448829] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[    4.303164] Loaded X.509 cert 'Build time autogenerated kernel key: 2eee0fc327b793a2ef723d47dfa1f1d192045836
'
[    4.308241] Key type encrypted registered
[    4.308305] ima: No TPM chip found, activating TPM-bypass! (rc=-19)
[    4.308424] evm: HMAC attrs: 0x1
[    4.357177] rtc-opal opal-rtc: setting system clock to 2018-01-13 07:00:20 UTC (1515826820)
[    4.359524] Freeing unused kernel memory: 11328K
[    4.376058] udevd[2639]: starting version 3.2.2
[    4.380417] udevd[2640]: starting eudev-3.2.2
[    4.425155] tg3 0005:01:00.0 enP5p1s0f0: renamed from eth0
[    4.425946] scsi 0:0:0:0: Attached scsi generic sg0 type 0
[    4.426109] scsi 1:0:0:0: Attached scsi generic sg1 type 0
[    4.430047] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    4.430080] sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    4.430083] sd 1:0:0:0: [sdb] 4096-byte physical blocks
[    4.430105] sd 1:0:0:0: [sdb] Write Protect is off
[    4.430144] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.430680] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    4.430765] sd 0:0:0:0: [sda] Write Protect is off
[    4.430903] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.433216]  sda: sda1 sda2 sda3
[    4.433863] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   89.846182] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   89.879775] systemd-shutdown[1]: Unmounting file systems.
[   90.143992] systemd-shutdown[1]: All filesystems unmounted.
[   90.144026] systemd-shutdown[1]: Deactivating swaps.
[   90.144169] systemd-shutdown[1]: All swaps deactivated.
[   90.144213] systemd-shutdown[1]: Detaching loop devices.
[   90.145648] systemd-shutdown[1]: All loop devices detached.
[   90.145679] systemd-shutdown[1]: Detaching DM devices.
[   90.146001] systemd-shutdown[1]: Detaching DM 253:2.
[   90.454327] shutdown: 7 output lines suppressed due to ratelimiting
[   90.494066] dracut Warning: Killing all remaining processes
dracut Warning: Killing all remaining processes
[   90.653434] XFS (dm-0): Unmounting Filesystem
[   90.690465] dracut Warning: Unmounted /oldroot.
[   90.709953] dracut: Disassembling device-mapper devices
Powering off.
[   90.788553] sd 1:0:0:0: [sdb] Synchronizing SCSI cache
[   90.789070] sd 1:0:0:0: [sdb] Stopping disk
[   91.189127] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[   91.189900] sd 0:0:0:0: [sda] Stopping disk
[   91.975693] tg3 0005:01:00.1: Refused to change power state, currently in D0
[   92.087451] reboot: Power down
[  141.309126920,5] OPAL: Shutdown request type 0x0...
 ```

sivassrr, 2018-01-16:
```root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r28-0-gc9e2c09"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r28"
root@witherspoon:# cat /media/pnor-ro-7617ca43/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.113
	op-build-v1.19-320-g2364968-dirty
	buildroot-2017.08-8-g5e23247
	skiboot-v5.9.8
	hostboot-7050d0a
	linux-4.13.16-openpower1-pf94f2fe
	petitboot-v1.6.3-pc7918e8
	machine-xml-ef5f800
	occ-39d5490
	hostboot-binaries-e6e59b3
	capp-ucode-p9-dd2-v2
	sbe-b6b7bb7```

sivassrr, 2018-01-16:
```root@witherspoon:# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:#```

sivassrr, 2018-01-16:
[journalctlwsbmc004warmbootfails.txt](https://github.com/openbmc/openbmc/files/1635906/journalctlwsbmc004warmbootfails.txt)

rfrandse, 2018-02-07:
<https://gerrit.openbmc-project.xyz/8899> Warm Reboot should call the host reboot object
Resolves  #2797 GUI: Warmboot fails to restart

closed 2018-02-12 by geissonator