openbmc/openbmc #2798: MAX31785revA should report each tach in the same sysfs entry

msbarth, 2018-01-16:
An update is needed in the use of the max31785 TACHSEL across each max31785 revision variant to ensure the revA reports each fan rotor in a dual rotor fan assembly in the proper sysfs entries consistently.

msbarth, 2018-01-23:
MAX31785revA rotor feedbacks appear to be in a slow/fast relationship in the READ_FAN_SPEED_1 register, therefore the fan*_0 object will always represent the slowest rotor (may or may not be the front) and marked non-functional first. Since the MAX31785revA latches its TACHSEL = 0 during a speed transition, no rotor(or fan) should have its functional state updated during a speed transition.

- [ ] Include all rotors within a trust group and determine a transition is occuring by all _1 rotors = 0 tach
- [ ] Stop all timers while transitioning to a target RPM(store current timer and restore after target reached?)
- [ ] Update fan*_0 object's functional state based on the first set of fan*_input sysfs files
- [ ] The fan*_1 object's functional state is based on the fast rotor, therefore to be faulted fan*_0 must be if a faulted state first (due to comparing target & feedback speeds when not transitioning)

rfrandse, 2018-02-21:
<https://gerrit.openbmc-project.xyz/9084> wspoon: MAX31785revA fan controller workaround
Resolves  #2798 MAX31785revA is reporting faulted rear rotor in different words

closed 2018-03-07 by geissonator