openbmc/openbmc #2799: KVM over IP Support

geissonator, 2018-01-16:
@jk-ozlabs to add some details.  Going to use VNC, need some data compression, and some kernel work.

jk-ozlabs, 2018-01-16:
To elaborate, here are the components that I'd think we'd need to support this (as we'd discussed last week):

 1. A method of pulling host video pixel data out of the memory-mapped VGA region of the BMC. This would involve a kernel driver, preferably exposing that data over a pre-existing interface
 1. A USB gadget driver to expose HID endpoints, to allow sending input events to the host
 1. A userspace daemon to:
    1. grab pixel data from the BMC kernel
    2. send input events to the USB HID gadget drivers
    3. authenticate incoming connections from a websocket connection
    4. send pixel data over the socket, probably using VNC as an existing protocol, and probably using libvnc to do that
    5. receive input events from the socket, and marshal those to the HID gadget drivers
 1. Implementing a javascript-based VNC client; I believe that there are existing open source implementations, but we'd probably need to make some modifications to work with our socket implementation.
 1. Modifications to the web UI to serve the javascript-based VNC client, to connect to the daemon above
 1. Infrastructure in the web UI mechanism to allow an authentication credentials from the existing logged-in session to be used for the websocket

jk-ozlabs, 2018-01-16:
FWIW, @edtanous already has some code to implement at least part of this. Ed, can you share what you've done so far?

edtanous, 2018-01-17:
Let me set expectations very clearly here: This was a proof of concept that was very limited in its capabilities and standards compliance.  It did have some greater thought put into the components selected for longer term viability, but it's in no way complete.  It managed about 1 frame per 5 seconds with an AST2500 platform, as all decoding was done on the arm, and not in the browser.

All parts of the POC are open (if you need any pieces that aren't, let me know)  It will need to be pieced together again for openbmc, as it was using very different build systems, a different web front end, and Kernel 3.18.
The linux driver for 3.18 is here:
https://downloadcenter.intel.com      /download/26968/BMC-Source-Code-for-Intel-62X-Series-Server-Boards-and-Systems?product=8901
S2600_BMC_Open_Src_Code-v1.03\S2600_OSS\IKVM\Driver\Video\OS\Li      nux\Host\AST2500\video_drv.c
I do have a modification of that driver targeted against 4.9 that should be open sourced shortly (I'll check on that status tomorrow).  This driver is far from upstream worthy, and needs some love.  In my opinion it should also use the kernels new contiguous memory allocator instead of running from unmapped RAM and have the ability to drop the malloced buffers when the user disconnects.  It also needs several general cleanups, and someone needs to invest some time to see if it could match the video 4 linux interfaces.

The KVM userspace implementation is below, and exposes an authenticated (using pam) VNC over websocket implementation.  At this point, it's very naive;  Opens the video device, grabs the frame buffer, and decodes it from the jpeg-like aspeed buffer to a bitmap, then sends it as the bitmap type to VNC.  Client VNC events are coalesced and handled by the switch statement on line 266.
https://gerrit.openbmc-project.xyz/#/c/7786/2/bmcweb/include/web_kvm.hpp

On the client side, I used stock novnc (https://github.com/novnc/noVNC) loaded out of an angular webui similar to the architecture of phosphor-webui.

The important thing to recognize is that to have the best performance pixel transformation needs to be pushed as close to the user as is possible.  Most of the hardware in use (both Nuvoton and Aspeed) support compression mechanisms, as well as screen difference algorithms in hardware, but to my knowledge neither is capable of producing standards compliant interfaces (Jpeg x264, mjpeg, ect) and generally produce SOC or vendor specific things.  This most likely means we need to either decode on the bmc (and take the terrible performance hit) or define new buffer types on top of an already existing protocol spec.

If you'd like more of my help in real time, I'm on IRC as well as on the Monday technical calls.

jk-ozlabs, 2018-01-17:
Hi Ed, that's fine - you made it clear that you didn't have a finished implementation, I'm more intending to avoid duplicating work with what you've already done. Thanks for the info, I'll take a look.

edtanous, 2018-04-28:
FYI, the beginnings of the HTML5 integration are staged here.
https://gerrit.openbmc-project.xyz/#/c/10268/

rfrandse, 2018-08-16:
moved to https://github.com/ibm-openbmc/dev/issues/43

closed 2018-08-16 by rfrandse