openbmc/openbmc #2800: SNMP Support in OpenBMC

geissonator, 2018-01-16:
Need to investigate what we have and what is available.   There was talk that we did some light version @ratagupt @bradbishop ?

geissonator, 2018-01-16:
Once initial work done we'll need to size it up and pick a milestone.

jpmulhol-ibm, 2018-02-06:
We have customers asking for SNMP trap functions.  They had this available on p8 BMC based systems and see this as a loss of function between p8 and p9 openBMC.  Commenting to ensure that traps are included in SNMP functionality.

AlexanderAmelkin, 2018-03-27:
I believe this issue has to be linked to #3002, #3004, and #3005.
Please take a look at https://github.com/YADRO-KNS/phosphor-snmp

closed 2018-08-17 by rfrandse

liudubang, 2020-07-09:
I just know how to config in the web ,but don't know how to use in other pc get a trap, can you give me some example,thanks