openbmc/openbmc #2801: Handle failure to update ubiblock

geissonator, 2018-01-16:
CI system hit this today.   We ended up in uboot.  @anoo1 looked at it and found we'd had an error doing the update, but still proceeded to try and reboot into it.  This issue is to make sure we handle this situation better.
```
Jan 16 21:11:09 witherspoon kernel: block ubiblock4_0: created from ubi4:0(rofs-91e149b6)
Jan 16 21:11:09 witherspoon kernel: blk_update_request: I/O error, dev ubiblock4_0, sector 0
Jan 16 21:11:09 witherspoon obmc-flash-bmc[1030]: mount: /dev/ubiblock4_0: can't read superblock

Volume ID:   0 (on ubi4)
Type:        static
Alignment:   1
Size:        257 LEBs (16809856 bytes, 16.0 MiB)
Data bytes:  0 bytes <=== (BAD)
```

anoo1, 2018-01-16:
Attaching the journal log.
[journalct.txt](https://github.com/openbmc/openbmc/files/1636927/journalct.txt)

gtmills, 2018-01-23:
going to add in `set -eo pipefail` to 
https://github.com/openbmc/openbmc/blob/master/meta-phosphor/common/recipes-phosphor/flash/phosphor-software-manager/obmc-flash-bmc since this problem is due to us, not detecting a failure in obmc-flash-bmc

rfrandse, 2018-01-23:
<https://gerrit.openbmc-project.xyz/8586> Add "set -eo pipefail" to obmc-flash-bmc
Resolves  #2801 Handle failure to update ubiblock

amboar, 2018-02-04:
@rfrandse I wouldn't say that patch resolves this issue. @gtmills do we have any insight into what the root cause of the `obmc-flash-bmc` failure is? If so can you post evidence here?

gtmills, 2018-02-05:
@amboar This issue is to handle the above fail better. 
We had a fail in activating the image and never detected it. We should have never tried to reboot into this image.
By using `set -eo pipefail`, https://gerrit.openbmc-project.xyz/#/c/8586/, we would now detect this fail, and set the Activation to failed. 
In the journal is 
`Jan 16 20:30:26 witherspoon obmc-flash-bmc[3550]: Unable to update volume rofs-91e149b6!`. We should have failed there but did not. 
In obmc-flash-bmc, ubiro() calls ubi_updatevol() but never checks the return code.. 
A change could have been `ubi_updatevol || exit 1` but I feel this change to use `set -eo pipefail` is better since we need to check the rc in more places than just this.

amboar, 2018-02-05:
Right, but why did we fail to update the rofs?

gtmills, 2018-02-05:
It failed because the file was 20MB instead of the allocated 16MB, but that is a different issue.
This issue is to handle the failure of ubiblock/ubi_updatevol. This issue is "we had a failure and did not detect it".

amboar, 2018-02-05:
Ok. Thanks for clarifying.

closed 2018-03-01 by geissonator