openbmc/openbmc #2803: GUI: Orderly shutdown & Immediate shutdown not working - System at OS after that operation

sivassrr, 2018-01-17:
Problem Description:

System was at running state (OS was up and running ), Did Orderly shutdown from GUI, But system failed to power off. 
Similarly tried Immediate shutdown from GUI, System failed to power off. 

Console Text:
Console indicates system is at OS state: ( Hit enter multiple times - OS prompt was keep coming - no sign of power off ) .
ws004os login:
Red Hat Enterprise Linux Server 7.4 Beta (Maipo)
Kernel 4.11.0-43.el7a.ppc64le on an ppc64le

ws004os login:
Red Hat Enterprise Linux Server 7.4 Beta (Maipo)
Kernel 4.11.0-43.el7a.ppc64le on an ppc64le

ws004os login:
Red Hat Enterprise Linux Server 7.4 Beta (Maipo)
Kernel 4.11.0-43.el7a.ppc64le on an ppc64le

ws004os login:
Red Hat Enterprise Linux Server 7.4 Beta (Maipo)
Kernel 4.11.0-43.el7a.ppc64le on an ppc64le

ws004os login:

============================================================
FW Details:
```
root@witherspoon:# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:#
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r28-0-gc9e2c09"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r28"
root@witherspoon:# cat /media/pnor-ro-7617ca43/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.113
	op-build-v1.19-320-g2364968-dirty
	buildroot-2017.08-8-g5e23247
	skiboot-v5.9.8
	hostboot-7050d0a
	linux-4.13.16-openpower1-pf94f2fe
	petitboot-v1.6.3-pc7918e8
	machine-xml-ef5f800
	occ-39d5490
	hostboot-binaries-e6e59b3
	capp-ucode-p9-dd2-v2
	sbe-b6b7bb7
root@witherspoon:# ```

sivassrr, 2018-01-17:
[journalctlwsbmc004immediateOrdershutdownfail.txt](https://github.com/openbmc/openbmc/files/1637720/journalctlwsbmc004immediateOrdershutdownfail.txt)

rfrandse, 2018-04-18:
<https://gerrit.openbmc-project.xyz/10204> Add fixes for power operations
Resolves  #2803 GUI: Orderly shutdown & Immediate shutdown not working - System at OS after that operation

closed 2018-04-26 by geissonator