openbmc/openbmc #2805: FVT:Witherspoon- OCC crashes during power on

cdeadmin, 2018-01-17:


geissonator, 2018-01-17:
Summary of Issue:

We're getting intermittent occ failures of the openpower-occ-control application when it's reading the OCC device.  We've seen in other applications, like with hwmon, that we just need to add some retries into the application layer for these.  With this issue, we're going to add some retriesd on the read call.  Here's some data from the fail:

```
Jan 15 08:48:03 witherspoon openpower-occ-control[1282]: Reading from OCC failed.
Jan 15 08:48:03 witherspoon openpower-occ-control[1282]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::ReadFailure'
Jan 15 08:48:03 witherspoon openpower-occ-control[1282]:   what():  org.open_power.OCC.Device.Error.ReadFailure: Reading from OCC failed.
Jan 15 08:48:03 witherspoon kernel: occ-hwmon occ1-dev0: occ bus op failed rc:-74
Jan 15 08:48:03 witherspoon kernel[1066]: [  331.580000] occ-hwmon occ1-dev0: occ bus op failed rc:-74
Jan 15 08:48:03 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Jan 15 08:48:03 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Jan 15 08:48:03 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'

```

rfrandse, 2018-01-17:
<https://gerrit.openbmc-project.xyz/8519> Retry OCC read operations
Resolves  #2805 FVT:System- OCC crashes during power on

closed 2018-01-24 by geissonator