openbmc/openbmc #2806: CORAL:OP910.10:STC Witherspoon xcat/rflash update of BMC failed on 1 of 9 Nodes

cdeadmin, 2018-01-17:


gtmills, 2018-01-17:
After freeSpace()  (https://github.com/openbmc/openpower-pnor-code-mgmt/blob/master/item_updater.cpp#L547) runs there is no guarantee the number of PNOR versions is at ACTIVE_PNOR_MAX_ALLOWED -1.  Currently, ACTIVE_PNOR_MAX_ALLOWED is
set to 2. We have seen cases where there are 3 active PNOR versions. In these cases when freeSpace() is called, the number of active PNOR versions only drops to 2, when it should drop to 1.

rfrandse, 2018-01-17:
<https://gerrit.openbmc-project.xyz/8516> fix item updater freeSpace()
Resolves  #2806 freeSpace() after running should bring the number of active PNOR versions to ACTIVE_PNOR_MAX_ALLOWED -1

closed 2018-01-22 by geissonator