openbmc/openbmc #2808: phosphor-fan: support fan pwm

mine260309, 2018-01-18:
Some systems (e.g. Romulus, Zaius) uses pwm to do fan control, instead of `fan_target` on Witherspoon.
The current phosphor-fan only supports `fan_target` variant to monitor and control fans.
This issue is to track the tasks for supporting fan pwm.

Currently, we have some work in progress:
1. Add Control.FanPwm interface: https://gerrit.openbmc-project.xyz/#/c/8357/
2. Add pwm as fan speed target: https://gerrit.openbmc-project.xyz/#/c/8353/

The remaining tasks may consists of:
1. Implement `FanPwm` interface in `phosphor-hwmon` to use this new interface:
   Current hwmon only support `FanSpeed` interface;
2. Support pwm in phosphor-fan-monitor:
   Current fan-monitor compare the fan speed with target value;
   For pwm, the fan speed is the rpm, while the target value is pwm, there shall be some conversion between rpm and pwm;
3. Some potential changes in phosphor-fan-control to support pwm.

pstrinkle, 2018-01-18:
Sorry for the delay, I was out of town and then swamped.  I have a "meh" implementation of the FanPwm within phosphor-hwmon.  I'd like to merge my patch with yours!  Because I like that your patch lets you assign the pwm to fan, and I like that mine uses a different interface.  Let me send that up now.

pstrinkle, 2018-01-18:
s/now/in a bit/ -- I have a patchset currently in review downstream that modernizes the code up to v2.0 of the tag, which will make sending it upstream easier.

mine260309, 2018-03-09:
This is completed by:

1. Add Control.FanPwm interface
   * https://gerrit.openbmc-project.xyz/#/c/8357/
2. Add pwm as fan speed target
   * https://gerrit.openbmc-project.xyz/#/c/8353/
3. Implement FanPwm interface in phosphor-hwmon to use this new interface
   * https://gerrit.openbmc-project.xyz/#/c/8710/
4. Support pwm in phosphor-fan-monitor
   * https://gerrit.openbmc-project.xyz/#/c/8741/
   * https://gerrit.openbmc-project.xyz/#/c/8661/
5. Support pwm in phospohr-fan-control
   * https://gerrit.openbmc-project.xyz/#/c/8743/

closed 2018-03-09 by mine260309