openbmc/openbmc #2811: Host watchdog is enabled on power on request when host is already on

mine260309, 2018-01-19:
When host is powered on and fully booted, the host watchdog is started and disabled.

Then if user issues power on request, e.g. `obmcutil poweron`, the watchdog is enabled again.
At this time, no one is resetting it and eventually the watchdog fires and cause host to be reset.

The watchdog is expected not to be enabled when host is already on.

mine260309, 2018-01-19:
This issue is reported by Yadro on P8 system, and can be reproduced on Romulus, Witherspoon and possibly all other systems.

mine260309, 2018-01-19:
`obmc-enable-host-watchdog@.service` specifies:
```
ConditionPathExists=!/run/openbmc/host@%i-on
```

But the file `/run/openbmc/host@%i-on` is not created when host is on?

geissonator, 2018-01-19:
Any reason someone is doing the "obmcutil poweron" when the system is already booted?  Just a use case type test?  I think the right solution here is for obmc-enable-host-watchdog@.service to have "RemainAfterExit=yes".  That ensures we don't run it again until someone has powered off and stopped it.  

The /run/openbmc/host@%i-on is used just to handle BMC resets with the host up conditions.

mine260309, 2018-01-22:
> Any reason someone is doing the "obmcutil poweron" when the system is already booted? Just a use case type test?
Anyway you can not prevent a user trying to power on a machine that is already powered on, it should not cause the host to be reset.

The solution `RemainAfterExit=yes` does work, let's put the fix in the code.

mine260309, 2018-01-23:
https://gerrit.openbmc-project.xyz/#/c/8574/

rfrandse, 2018-01-23:
<https://gerrit.openbmc-project.xyz/8574> Fix host reset on double poweron command
Resolves  #2811 Host watchdog is enabled on power on request when host is already on

closed 2018-01-23 by geissonator