openbmc/openbmc #2813: Initial boot of BMC takes too long

madscientist159, 2018-01-21:
Any boot / reboot of the BMC (normally as a result of standby power removal) takes way too long, typically in excess of 1-2 minutes from standby power application to READY state / ability to IPL host.

U-boot itself consumes around 15 seconds of this with kernel load from Flash and subsequent verification, but the largest offender seems to be something in the systemd startup sequence.  Toward the end of the boot process the CPU is not the limiting factor, so there is a chance that some systemd dependency is timing out before the host can be IPLed.

This was first noted on the Talos and Romulus systems.

amboar, 2018-02-04:
@madscientist159 can you post some measurements using [`systemd-bootchart`](http://man7.org/linux/man-pages/man1/systemd-bootchart.1.html)?

madscientist159, 2018-02-05:
Absolutely!  Looks like most of the time is wasted waiting for the mapper service if I'm reading this correctly.
[image]

amboar, 2018-02-05:
Ok. There's a fairly ambiguous issue open about improving mapper performance in #2860. Hopefully @tomjoseph83 can provide more information on what his intent is there.

madscientist159, 2018-02-05:
Yeah, we had the BMC brick itself on one of our first boards at a customer site.  We're still waiting for a ROM dump to see what happened, but I'd guess right now that system power got switched off during the lengthy BMC boot and something got corrupted as a result.  The BMC should boot in under 30 seconds from where we sit, and ideally under 15 seconds.

madscientist159, 2018-02-23:
Recovery could be as simple as running this if the root mount fails:

```
cd /run/initramfs/rw/cow
ls | grep -v etc | grep -v var | xargs rm -rf
reboot -f
```

We'll probably add this as a failsafe in our builds.

geissonator, 2018-02-26:
I've been working this in parallel with some other issues dealing with performance on BMC resets.  The number 1 thing I've been working on has been removing the legacy python applications from our images.  A lot of these python applications from skeleton are no longer used, yet still loaded and started.

A few commits have made it in:
https://gerrit.openbmc-project.xyz/#/c/8934/ (UBI only enhancement)
https://gerrit.openbmc-project.xyz/#/c/8874/ (system manager legacy function removal)
https://gerrit.openbmc-project.xyz/#/c/8871/ (legacy settings removal)
https://gerrit.openbmc-project.xyz/#/c/8834/ (legacy system state removal)

Some have not due to small uses in other repos (or waiting for merge):
https://gerrit.openbmc-project.xyz/#/c/9055/ (netman.py removal)
https://gerrit.openbmc-project.xyz/#/c/8829/ (settings removal)
https://gerrit.openbmc-project.xyz/#/c/8793/ (inventory removal)
https://gerrit.openbmc-project.xyz/#/c/9004/ (BMC control removal)

With the above changes I managed to reduce our boot time by about 20%. There's still some skeleton python applications that need to be refactored and moved to c++ function (chassis_control.py, system_manager.py, sensor_manager2.py, sync_inventory_items.py). 

The other thing we've looked at is refactoring mapper into a C++ application.  I believe @edtanous has a prototype for this and the initial results looked promising.  Mapper by it's nature has a ton of processing to do because it has to introspect every existing dbus service when it starts and every new one that comes on the bus, so it's the number 1 user of CPU during a BMC boot.

edtanous, 2018-05-04:
FYI, prototype is now available for experimenting.  It has a ways to go before it's merge worthy, but should be able to validate some assumptions about the path we're going down.

https://gerrit.openbmc-project.xyz/#/c/10468/
https://gerrit.openbmc-project.xyz/#/c/10470/

amboar, 2018-05-10:
In the mean time, I've also got some very much WIP patches to improve the performance of the current mapper implementation:

https://gerrit.openbmc-project.xyz/#/q/topic:mapper-perf

geissonator, 2018-07-12:
Some more good stuff that got us another 10 seconds on the reboot:
https://gerrit.openbmc-project.xyz/#/c/11402/
https://gerrit.openbmc-project.xyz/#/c/11427/

madscientist159, 2019-04-27:
This is still a significant concert on the latest OpenBMC master.

gtmills, 2019-11-16:
@madscientist159 Still a problem? Still want this issue open?

classilla, 2019-11-16:
I'm not him, but I'm curious about the improvements. Which commit(s)?

geissonator, 2019-11-19:
We have removed all python from our image, moved to the c++ mapper and c++ bmcweb REST server. All of this helped....but we have also continued to add new features and services that must be started. So, with master, on our witherspoon here is where we're at:

```
2.8.0-dev-877-g9d570fc

Nov 19 15:14:50 witherspoon-Y230UF71K03T systemd[1]: Startup finished in 8.472s (kernel) + 1min 44.506s (userspace) = 1min 52.978s.
```
I have noticed on the AST2600 evb that this same code runs 3-4x faster so that will be a nice upgrade when we get to it. But for AST2500, I'm not sure there's much more low hanging fruit.

gtmills, 2019-11-19:
Moving to dbus-broker helped in this area as well. 
https://github.com/openbmc/openbmc/commit/bba6b5a88485e9101b83bf46b846509751d439be

classilla, 2019-11-19:
Very interesting. Thank you.

amboar, 2019-11-20:
I'd argue that 1m52s is still to slow:

>  The BMC should boot in under 30 seconds from where we sit, and ideally under 15 seconds.

gtmills, 2019-11-20:
> I'd argue that 1m52s is still to slow

Reopened

closed 2020-06-21 by stale[bot]