openbmc/openbmc #2820: Code update test fails with v2.0-199-g125f40a != v2.0-202-gb100912

geissonator, 2018-01-23:
Seen this a few times on the CI machine.  I thought we had another issue tracking this but can't find it.  Basically we do the update and do the BMC reboot but we don't end up on the correct image.

```
/xyz/openbmc_project/software/e97d5602
{
    "associations": [
        [
            "inventory", 
            "activation", 
            "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
    ], 
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
    "Priority": 1, 
    "Version": "v2.0-202-gb100912", 
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
    "Path": "", 
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None"
}
/xyz/openbmc_project/software/f0315569
{
    "associations": [
        [
            "inventory", 
            "activation", 
            "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
    ], 
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active", 
    "Priority": 0, 
    "Version": "v2.0-199-g125f40a", 
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC", 
    "Path": "", 
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None"
}
```
```
Jan 23 13:04:24 witherspoon systemd[1]: Started Phosphor Settings Manager.
Jan 23 13:04:24 witherspoon systemd[1]: Starting Wait for /org/openbmc/settings/host0...
Jan 23 13:04:25 witherspoon systemd[1]: Started Read OpenPOWER-format VPD from EEPROM.
Jan 23 13:04:25 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/uuid...
Jan 23 13:04:25 witherspoon systemd[1]: Starting OpenBMC Software Update Manager...
Jan 23 13:04:32 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 23 13:04:32 witherspoon systemd[1]: Starting Set U-Boot environment variable...
Jan 23 13:04:32 witherspoon systemd[1]: Started OpenBMC Software Update Manager.
Jan 23 13:04:32 witherspoon systemd[1]: Starting Phosphor Version Software Manager...
Jan 23 13:04:33 witherspoon systemd[1]: Started Phosphor Version Software Manager.
Jan 23 13:04:35 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 23 13:04:35 witherspoon systemd[1]: Started Sync system/chassis/motherboard/uuid.
Jan 23 13:04:36 witherspoon systemd[1]: Started Set U-Boot environment variable.
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]: Traceback (most recent call last):
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:     result = self._run(*self.args, **self.kwargs)
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:     ciphers=ciphers)
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:     self.do_handshake()
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]:     return self._sslobj.do_handshake()
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Jan 23 13:04:36 witherspoon phosphor-gevent[1180]: <Greenlet at 0x7609ae40: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x75bfaeb0 fileno=6 address=('::3472:f776:d85d:6d76%1986878920', 443, 0, 1986878920)>>(<socket at 0x75b39c70 fileno=10>, ('::c059:6b76:8e91:f402%1995856668', 38332, 0, 199)> failed with SSLEOFError
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]: Traceback (most recent call last):
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     result = self._run(*self.args, **self.kwargs)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     ciphers=ciphers)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     self.do_handshake()
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     return self._sslobj.do_handshake()
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]: <Greenlet at 0x7609ae90: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x75bfaeb0 fileno=6 address=('::3472:f776:d85d:6d76%1986878920', 443, 0, 1986878920)>>(<socket at 0x75b39c10 fileno=11>, ('::c059:6b76:8e91:f402%1995856668', 38334, 0, 199)> failed with SSLEOFError
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]: Traceback (most recent call last):
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     result = self._run(*self.args, **self.kwargs)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     ciphers=ciphers)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     self.do_handshake()
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]:     return self._sslobj.do_handshake()
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Jan 23 13:04:37 witherspoon phosphor-gevent[1180]: <Greenlet at 0x7609aee0: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x75bfaeb0 fileno=6 address=('::3472:f776:d85d:6d76%1986878920', 443, 0, 1986878920)>>(<socket at 0x75b39cb0 fileno=12>, ('::c059:6b76:8e91:f402%1995856668', 38336, 0, 199)> failed with SSLEOFError
Jan 23 13:04:38 witherspoon avahi-daemon[988]: Service name conflict for "phosphor_rest" (/services/phosphor_rest.service), retrying with "phosphor_rest #17".
Jan 23 13:04:38 witherspoon avahi-daemon[988]: Service name conflict for "obmc_console" (/services/obmc_console.service), retrying with "obmc_console #17".
Jan 23 13:04:39 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/network/config.

```

```
[witherspoon]# fw_printenv | grep -e "=0" -e "=1"
6ff51f1b=1
baudrate=115200
bootargs=console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=0,0 root=/dev/ubiblock0_0
f9753315=1
fba43456=1
mtddevnum=0
pnor-7a9ef57c=1
pnor-8764baa3=0
e97d5602=1
f0315569=0

```

geissonator, 2018-01-25:
Hit again today on a CT job:
```
v2.0-219-g07d198c != v2.0-214-gca5a08f
```

anoo1, 2018-02-27:
Fixed with https://github.com/openbmc/openbmc/issues/2764

closed 2018-02-27 by anoo1