openbmc/openbmc #2822: host watchdog service error during boot

geissonator, 2018-01-23:
Seeing this in master on witherspoon.  System still boots fine but not sure what's up with this error.

```
Jan 23 20:48:16 witherspoon mapper[3465]: Error finding '/xyz/openbmc_project/watchdog/host0' service: No such file or directory
Jan 23 20:48:16 witherspoon systemd[1]: obmc-enable-host-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Jan 23 20:48:16 witherspoon systemd[1]: Failed to start Start Watchdog0.
Jan 23 20:48:16 witherspoon systemd[1]: obmc-enable-host-watchdog@0.service: Unit entered failed state.
Jan 23 20:48:16 witherspoon systemd[1]: obmc-enable-host-watchdog@0.service: Failed with result 'exit-code'.
Jan 23 20:48:16 witherspoon systemd[1]: Reached target Start Host0.
```

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.0-199"
VERSION_ID="v2.0-199-g110bec8"
```

mine260309, 2018-01-24:
This seems to indicate that the host watchdog is not started yet, since the `obmc-enable-host-watchdog@0.service` is after `mapper-wait@-xyz-openbmc_project-watchdog-host%i.service`.

What's the full journal log of this issue?

geissonator, 2018-06-04:
This seems to have cleaned itself up.

closed 2018-06-04 by geissonator