openbmc/openbmc #2824: Fan Control: Control mode should set fans to a given speed upon starting

msbarth, 2018-01-24:
In the case where the BMC is rebooted while the host is running, the fan control application (control mode) should be able to be configured to set the target to a given speed.

amboar, 2018-02-04:
Isn't it better if we don't do anything specific, and just let the fan control algorithm automatically adjust the fan speed as a function of system environmentals? Why do we need to do anything specific in the start-up path?

msbarth, 2018-02-12:
As noted, in the case where the the BMC is rebooted (or some other reason that the fan control application is restarted...i.e. during thermal characterization) upon the fan control application (control mode) is started, we want it to be able to set the target to a given defined speed.
This will be done thru configuring a set speed event of some sort that will set the target speed and update the applications internal target speed variable to some value. The value, for example, can be hard coded by the user...or can be a function to read the last set target speed property and use that value.

amboar, 2018-02-12:
Sure, but it's still not clear to me why we would want to do that independent of the system environmentals? If you just want some mechanism to control the fan target, you can use the sysfs interface directly right?

msbarth, 2018-02-13:
I dont understand where the confusion is...this change has nothing to do with system environmentals, strictly on addressing the issue where at anytime the fan control application is stopped/started unexpectedly that upon starting up it correctly sets the target speed (internally & externally) using some user defined action.
One defined action case may be to read the target speed property (which was the last set target speed before the fan control application was restarted) and update its internal cache of the target speed to that value and then re-write it out to hwmon. This is to minimize fan oscillations, etc.. from happening while the fan control application may be restarting multiple times (by systemd) or by a thermal engineer user performing thermal characterizations where they do not want the fan control algorithm active, but then restart the service when finished.

amboar, 2018-02-16:
Sorry, I think I misunderstood the intent here.

msbarth, 2019-04-04:
This is still necessary as an available configuration for phosphor-fan-presence/control.

closed 2019-11-03 by stale[bot]