openbmc/openbmc #2825: IPL: host and chassis off after BMC reset from runtime on wsbmc016 with 910.1746.20171213a

cdeadmin, 2018-01-24:


geissonator, 2018-01-24:
We have a CPU overload issue when the BMC is first starting all of it's services.  This can cause mapper related timeouts, which if they hit the right (or wrong depending on how you look at it) service then the BMC can end up in a bad state.  Mapper has been tuned about as much as it can for now but there's a lot of legacy python applications getting loaded that for the most part are not needed.  Need to get rid of these, but for now just make them lower priority.

rfrandse, 2018-01-25:
<https://gerrit.openbmc-project.xyz/8627> Run legacy services with lowest priority
Resolves  #2825 host and chassis off after BMC reset from runtime on with latest code

closed 2018-02-08 by geissonator