openbmc/openbmc #2826: Delete systemd-coredump temporary files from flash during boot.

ojayanth, 2018-01-25:
In case systemd-coredumps dies during core file processing,  /var/lib/systemd/coredump/.#core... stays on the flash and existing systemd-tmpfiles doesn't remove /var/lib/systemd/coredump/.#core... on boot. 

This can potentially lead’s the filesystem full scenario and related issues 

Need to provide better mechanism to do clean-up these files during boot.

ojayanth, 2018-02-07:
- [ ] add rules in tmpfiles.d to delete /var/lib/systemd/coredump/.#core... .

- [ ] Make sure that systemd-coredump process starts only after this tmpfiles based clean-up. This will help to reduce complexity of  handling  the active core processing  in the current boot path.

- [ ] Stop sytemd-coredump service in the beginning of shutdown process , This will help to reduce the chance of creating multiple  cores in the failed path shutdown -- This need more discussion.

There is already an issue https://github.com/openbmc/openbmc/issues/2538 to reduce the total core size to 500KB.

rfrandse, 2018-04-03:
<https://gerrit.openbmc-project.xyz/9156> Remove hidden-files in coredump(dir) during boot
Partially Resolves  #2826 Delete systemd-coredump temporary files from flash during boot.

rfrandse, 2018-09-04:
moved to https://github.com/ibm-openbmc/dev/issues/152

closed 2018-09-04 by rfrandse