openbmc/openbmc #2831: SBE update IPL loop

madscientist159, 2018-01-26:
With a relatively recent OpenBMC build, if hostboot is updated and needs to update the SBE, the system will enter a loop where it continually retries IPL and SBE update.  This is caused by the processor not even attempting to boot from SBE side 0 after the re-IPL is requested after SBE side 0 update.

Rebooting the BMC after the SBE update completes resolves the loop.  Standby power does not need to be removed for this to work.

Filing this against OpenBMC since I suspect it is simply requesting IPL from the wrong SBE side after a re-IPL request.

madscientist159, 2018-01-26:
Note that we're using our own PNOR builds from here:

https://git.raptorcs.com/git/talos-op-build/

The critical part is probably the hostboot version, which is https://github.com/open-power/hostboot/commit/cc2d45afa61c873f36d04aa5e4b095321ea677d1

k3a, 2018-07-02:
Experienced the infinite update loop as well - here is the hostboot log https://gist.github.com/k3a/1978de7ffb875e034fb35bf509bded3d. Removing power completely, waiting and booting again resulted in a successful boot.

```
15.33271|Error reported by sbe (0x2200) PLID 0x9000000A
15.33636|  <none>
15.33636|  ModuleId   0x0a unknown
15.33637| ReasonCode 0x221c unknown
```
According to https://github.com/open-power/hostboot/blob/ca30e6bf9ea440ba1fd8c7fb5092b9fe6e18aba0/src/include/usr/sbe/sbereasoncodes.H: 

```
SBE_DECISION_TREE = 0x0A,
SBE_BOOT_SIDE_DIRTY_BAD_PATH = SBE_COMP_ID | 0x1C,
````

```
//// 0x06: side0=clean, side1=dirty, boot=1 ///////////////////////
// Uncommon situation: booting from dirty side after other side was updated
```

geissonator, 2018-07-12:
We definitely know the SBE update paths works, it's used all the time as new images are released on our witherspoon systems.  There were a few bugs in this path which I could see getting hit when this bug was initially opened.

That log from @k3a above doesn't really show me an error though on the BMC side?  Hostboot has 2 spots they can update the SBE.  Looks like they needed to update in both in the above data.

The  BMC logic is very simple when selecting a boot side. There's a counter that starts at 3, it will use side 0 until the counter reaches 1 at which point it will switch over to side 1.  When hostboot is doing updates, it resets this counter to 3, to ensure it's booting from side 0.

madscientist159, 2018-07-13:
We've narrowed the fault down to a path that is run when the nest configuration changes.  When the SBE is updated right after ISTEP 7,3 (e.g. to apply a different nest frequency than the one previously used by the SBE), two reboots are required afterward (ISTEP 10,2 then again to boot into the new SBE image).  Instead of booting from side 0 after the second SBE update, we boot into side 1, then enter a loop.

closed 2019-04-26 by stale[bot]