openbmc/openbmc #2832: IPL: host off / chassis on after BMC reset from runtime on w51 with 910.1746.20180109a

cdeadmin, 2018-01-26:


geissonator, 2018-01-26:
OpenBMC is dealing with some performance issues right after
    a reboot. Systemd does almost too good a job at parallelizing
    all of the services which causes a 6X hit on the processor
    during the initial boot once mapper is ready.
    
    In situations where the BMC is rebooted when the host is up,
    this performance issue coupled with errors queued up by the
    host to send down to the BMC, can cause OpenBMC to exceed
    its current 3 second timeout for attentions.
    
    This initial message from OpenBMC to the host is done to
    check if the host is running. If the message times out then
    it's assumed the host is not running which causes all sorts

My proposal for now is to just bump up the timeout that we give the host to respond.

rfrandse, 2018-01-26:
<https://gerrit.openbmc-project.xyz/8679> Increase timeout for host to respond to attentions
Resolves  #2832  host off / chassis on after BMC reset from runtime on witherspoon system

closed 2018-01-29 by geissonator