openbmc/openbmc #2835: Image Signing: Create private/public key pair

anoo1, 2018-01-28:


amboar, 2018-01-29:
Are you suggesting we have a unique key for every build? My gut feeling would be to go the other way: Provide infrastructure for people to set one or more keys to be used for signing, and provide a publicly exposed keypair that's configured by default. That way it better aligns with how those publishing images would use the system (one signing key that's used repeatedly).

anoo1, 2018-01-31:
Yeah, I meant to describe what you said, to have re-usable keys.

amboar, 2018-02-02:
As pointed out by Vernon on the mailing list, having a default key is dangerous because someone will deploy with it and compromise any security offered by image signing for their platform. The reason I suggested it was so that we were always exercising the image signing path in the code when testing, even with, say, CI images. Maybe we can have a CI-specific key or something? Maybe it's just not worth the hassle given the risk?

rfrandse, 2018-03-08:
<https://gerrit.openbmc-project.xyz/8949> Add image signing framework and open keys
Resolves  #2835 Image Signing: Create private/public key pair

closed 2018-03-30 by geissonator