openbmc/openbmc #2836: Image Signing: Sign image during build

anoo1, 2018-01-28:


amboar, 2018-01-29:
I've started looking around at whether there are existing mechanisms for this. So far I've come across [swupdate](https://github.com/sbabic/swupdate) and its [Yocto layer](https://github.com/sbabic/meta-swupdate) but not much else.

It looks like swupdate might be a bit of a drastic change just to get image signing, but I'll keep digging in case there's a clean way to integrate it into OpenBMC.

anoo1, 2018-02-08:
We can use the swupdate way of signing images based on their bbclass: https://github.com/sbabic/meta-swupdate/blob/5a169fbaa5945c74f1d0d9e3f1f9da82130fa0f1/classes/swupdate.bbclass

rfrandse, 2018-03-08:
<https://gerrit.openbmc-project.xyz/8949> Add image signing framework and open keys
Resolves  #2836 Image Signing: Sign image during build

closed 2018-03-30 by geissonator