openbmc/openbmc #2842: Set BMC static ip from restful api

ryandcyu, 2018-01-29:
@anoo1 @mine260309 

I have added an one static ip through 
curl -b cjar -k -X GET  https://192.168.1.130//xyz/openbmc_project/network/eth0/ipv4/c4adbd86
{
  "data": {
    "Address": "192.168.1.130",
    "Gateway": "192.168.1.1",
    "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
    "PrefixLength": 24,
    "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
  },
  "message": "200 OK",
  "status": "ok"
}
Add I modified the Address property through
curl -b cjar -k -X PUT -H "Content-Type:application/json" -d '{"data": "192.168.1.131"}' https://192.168.1.130/xyz/openbmc_project/network/eth0/ipv4/c4adbd86/attr/Address

Then it seems workable

curl -b cjar -k -XGET  https://192.168.1.130//xyz/openbmc_project/network/eth0/ipv4/c4adbd86                                          {
  "data": {
    "Address": "192.168.1.131 ",
    "Gateway": "192.168.1.1",
    "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
    "PrefixLength": 24,
    "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
  },
  "message": "200 OK",
  "status": "ok"

But actually, BMC did not change the static ip. BMC was still keeping original ip.
ip addr:
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:1f:e2:fe:03:07 brd ff:ff:ff:ff:ff:ff
    inet 169.254.86.197/16 brd 169.254.255.255 scope link eth0
       valid_lft forever preferred_lft forever
    inet 192.168.1.130/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::21f:e2ff:fefe:307/64 scope link
       valid_lft forever preferred_lft forever

Is it a right way to modified ip of BMC? Or just only using POST method to create a new one ip and using DELETE method to delete the old ones?

mine260309, 2018-01-29:
Is it possible to make such properties read-only?

anoo1, 2018-01-30:
hi @ratagupt , could you look at this issue?

ratagupt, 2018-01-30:
@ryandcyu :- Currently property update is not supported,What you can do is delete the ipv4 object and add the new ip object with the changed properties.
We have some other story where we should not return the 200 OK for property update.

ryandcyu, 2018-01-30:
@ratagupt , Got it

closed 2018-01-30 by ryandcyu