openbmc/openbmc #2843: OpenBMC GUI - When system state is in Unreachable, there is no option under "Select a power operation"

samahaba, 2018-01-29:
[image]

karlvochatzer, 2018-02-20:
Please handle this situation by displaying the following message under the "Select a power operation" section:
“There are no power operations to display while a power operation is in progress. When complete, any new power operations will be displayed here.”

rfrandse, 2018-02-23:
<https://gerrit.openbmc-project.xyz/9174> Sloving When system state is in Unreachable, there is no option under "Select a power operation issue
fixes  #2843 OpenBMC GUI - When system state is in Unreachable, there is no option under "Select a power operation"

closed 2018-03-07 by geissonator