openbmc/openbmc #2847: phosphor-watchdog writing a lot of data to the journal - watchdog: update time remaining

geissonator, 2018-01-29:
I noticed when https://gerrit.openbmc-project.xyz/#/c/8712/ came through CI, it picked up https://gerrit.openbmc-project.xyz/#/c/8605/ which added some additional debug to the watchdog.

On our witherspoon, the debug output really filled up our logs during boot a lot.  Especially later in the boot when the host is nearing the end of it's boot.

```
Jan 29 20:14:58 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:14:58 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:14:58 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:14:58 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:14:58 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:14:59 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:14:59 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:14:59 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:14:59 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:14:59 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:14:59 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:14:59 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:14:59 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:14:59 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:14:59 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:14:59 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:14:59 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:14:59 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:00 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:00 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:00 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:15:00 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:15:00 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:15:00 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:15:00 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:00 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:00 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:15:00 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:15:00 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:00 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:01 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon ipmid[1272]: No Registered handlers for NetFn:[0x32],Cmd:[0xFF]
Jan 29 20:15:02 witherspoon ipmid[1272]: ERROR:[0xC1] handling NetFn:[0x32], Cmd:[0xF0]
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:02 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:03 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:03 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:03 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:03 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:03 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:04 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:04 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:04 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:05 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:05 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:05 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
[... 74 lines cut ...]
Jan 29 20:15:23 witherspoon ipmid[1272]: Bus: :1.72, Path: /org/open_power/control/occ1, Interface: org.open_power.OCC.Status
Jan 29 20:15:25 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jan 29 20:15:27 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jan 29 20:15:28 witherspoon phosphor-fan-control[4170]: Failed to invoke DBus method.
Jan 29 20:15:28 witherspoon phosphor-fan-control[4170]: The operation failed internally.
Jan 29 20:15:28 witherspoon phosphor-fan-control[4170]: Property not used, unless PropertyChanged signal received
Jan 29 20:15:28 witherspoon phosphor-fan-control[4170]: Failed to invoke DBus method.
Jan 29 20:15:28 witherspoon phosphor-fan-control[4170]: The operation failed internally.
Jan 29 20:15:28 witherspoon phosphor-fan-control[4170]: Property not used, unless PropertyChanged signal received
Jan 29 20:15:29 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:29 witherspoon phosphor-fan-control[4170]: Failed to invoke DBus method.
Jan 29 20:15:29 witherspoon phosphor-fan-control[4170]: The operation failed internally.
Jan 29 20:15:30 witherspoon phosphor-fan-control[4170]: Property not used, unless PropertyChanged signal received
Jan 29 20:15:31 witherspoon phosphor-fan-control[4170]: Failed to invoke DBus method.
Jan 29 20:15:31 witherspoon phosphor-fan-control[4170]: The operation failed internally.
Jan 29 20:15:31 witherspoon phosphor-fan-control[4170]: Property not used, unless PropertyChanged signal received
Jan 29 20:15:31 witherspoon phosphor-watchdog[3706]: watchdog: update time remaining
Jan 29 20:15:32 witherspoon phosphor-fan-control[4170]: Failed to invoke DBus method.
Jan 29 20:15:32 witherspoon phosphor-fan-control[4170]: The operation failed internally.
Jan 29 20:15:32 witherspoon phosphor-fan-control[4170]: Property not used, unless PropertyChanged signal received
Jan 29 20:15:32 witherspoon phosphor-watchdog[3706]: watchdog: disabled

```

We could just disable "debug" but I know there's some other useful logs in debug right now that would need to probably be moved.

wak-google, 2018-01-29:
So is this a case of you wanted debug logging on but want to filter out or not debug log from the watchdog? It seems like this is intended behavior if debug logging is turned on for all daemons. You can use journalctl to filter by unit or by priority to remove the debug levels. It's nice to be able to see from the logs if the host is actually updating the watchdog when debugging watchdog related issues.

wak-google, 2018-01-29:
Also that looks like a lot of updates per second compared to what I would expect.

geissonator, 2018-01-29:
On witherspoon we write to a persistent journal in flash, which means we only have a total size of 2.5MB so it's not as much about filtering as it is about ensuring we get the most valuable data in those 2.5MB.  

I don't think we've been all that consistent with our use of journal log levels in OpenBMC.  I could just make our witherpoon systems journald level the one above debug, but then we'd have to scrub all the debug logs from other services that we actually want.

wak-google, 2018-01-29:
It's really too bad journald doesn't have the ability to deal with partitioned journals such that NOTICE or lower messages go to a persistent journal and DEBUG / INFO go to a memory backed journal.

geissonator, 2018-01-30:
So...what do you want to do here?  You indicated it looked like a bit more logs then you thought above?  Is there a happy medium where we can still log these but reduce them?  I'm currently holding off on merging all phosphor-watchdog.bb bumps on this one.

wak-google, 2018-01-30:
I'd say we just revert the debug logging commit for now since this seems like a much more involved issue to clean up:
  - Debug logging shouldn't be the default for a system or persisted to the flash
    - We need to audit what levels the servers emit logs at before we could change that level
    - We need a way to access debug logging info if we don't write it to the journal
  - The host side of the watchdog in your sample log seems to be pinging at an abnormally high rate although I don't know much about the configuration. It really shouldn't need to ping more than once per second.

geissonator, 2018-02-01:
Yeah, I agree, lets just revert for now.  I know our host code design is to reset the watchdog on every "istep" in the boot, some of which happen very quickly.

rfrandse, 2018-02-15:
Code was reverted

closed 2018-02-15 by rfrandse