openbmc/openbmc #2850: FVT: BMC reset realod -  Failed to start Reload mboxd during power on

cdeadmin, 2018-01-30:


geissonator, 2018-01-30:
Summary of issue:

The mboxd-reload@.service requires the DBUS object provided by mboxd.service but there's no explicit definition of this dependency in the service files.  In most cases things worked fine, but when the BMC is heavily loaded, the initial startup of mboxd can take a bit of time to map in PNOR and cause mboxctl to not find the dbus object when it starts.

Here's some trace showing the issue:
```
Jan 30 05:57:50 witherspoon systemd[1]: Started Phosphor MBOX Daemon.
...
Jan 30 05:57:53 witherspoon systemd[1]: Starting Reload mboxd during power on...
Jan 30 05:57:53 witherspoon systemd[1]: Starting Witherspoon Fan Watchdog Reset...
Jan 30 05:57:53 witherspoon systemd[1]: Created slice system-fsi\x2dbind.slice.
Jan 30 05:57:53 witherspoon systemd[1]: Created slice system-ir35221\x2don\x2dbind.slice.
Jan 30 05:57:54 witherspoon systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Jan 30 05:57:56 witherspoon mboxctl[1417]: Failed to post message: No route to host
Jan 30 05:57:56 witherspoon mboxctl[1417]: Failed to send reset command
Jan 30 05:57:54 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/led/groups/power_on.
Jan 30 05:57:54 witherspoon systemd[1]: mboxd-reload@0.service: Main process exited, code=exited, status=255/n/a
Jan 30 05:57:54 witherspoon systemd[1]: Failed to start Reload mboxd during power on.
Jan 30 05:57:54 witherspoon systemd[1]: mboxd-reload@0.service: Unit entered failed state.
Jan 30 05:57:54 witherspoon systemd[1]: mboxd-reload@0.service: Failed with result 'exit-code'.
```

geissonator, 2018-01-30:
FYI @amboar

rfrandse, 2018-01-30:
<https://gerrit.openbmc-project.xyz/8730> Don't run mbox reset until dbus object available
Resolves  #2850 FVT: BMC reset realod -  Failed to start Reload mboxd during power on

closed 2018-02-01 by geissonator