openbmc/openbmc #2851: FVT: phosphor-reboot-host@0.service: Main process exited, code=exited, status=1/FAILURE

cdeadmin, 2018-01-30:
<cde:info> SW415527 </cde:info>

geissonator, 2018-01-30:
Summary of issue:

We pulled in https://gerrit.openbmc-project.xyz/#/c/8577/, patch set 6, which caused a regression in our host reset path.  The new target, obmc-chassis-powered-off@.target, runs last is the power off path, so the host reset service needs to go after that now.  Put a fix up in patch set 7 - https://gerrit.openbmc-project.xyz/#/c/8577/7/meta-phosphor/common/recipes-phosphor/state/files/phosphor-reboot-host%2540.service

```
Jan 30 06:49:42 witherspoon phosphor-chassis-state-manager[1263]: Received signal that power OFF is complete
Jan 30 06:49:42 witherspoon phosphor-chassis-state-manager[1263]: Change to Chassis Power State
Jan 30 06:49:42 witherspoon phosphor-host-state-manager[1295]: Received signal that host is off
Jan 30 06:49:42 witherspoon phosphor-host-state-manager[1295]: Change to Host State
Jan 30 06:49:42 witherspoon phosphor-timemanager[1312]: Changing time settings allowed now
Jan 30 06:49:42 witherspoon power_control.exe[1261]: BootProgress: xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified
Jan 30 06:49:42 witherspoon systemctl[1835]: Job for obmc-host-start@0.target canceled.
Jan 30 06:49:42 witherspoon systemd[1]: phosphor-reboot-host@0.service: Main process exited, code=exited, status=1/FAILURE
Jan 30 06:49:42 witherspoon systemd[1]: phosphor-reboot-host@0.service: Unit entered failed state.
Jan 30 06:49:42 witherspoon systemd[1]: phosphor-reboot-host@0.service: Failed with result 'exit-code'.
Jan 30 06:49:42 witherspoon systemd[1]: Stopped target Fan Control Ready.
Jan 30 06:49:42 witherspoon systemd[1]: Stopping Phosphor Fan Control Daemon...
Jan 30 06:49:42 witherspoon systemd[1]: Stopping Phosphor Fan Monitor Daemon...
Jan 30 06:49:42 witherspoon systemd[1]: Reached target Chassis0 (Powered Off).
```

geissonator, 2018-07-12:
Fixed in patch set above.

closed 2018-07-12 by geissonator