openbmc/openbmc #2855: Process 1034 (openpower-occ-c) of user 0 dumped core

PavamanSubramaniyam, 2018-01-31:
Install a P9 Zaius Hardware with upstream Open BMC and PNOR images.
Then in the Open BMC journalctl logs seeing the OCC service dumping core.

```
root@zaius:~# journalctl  | grep -i "core-dump"
Jan 31 07:03:58 zaius systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Jan 31 07:24:45 zaius systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.

root@zaius:~# journalctl  | grep -i core
Jan 31 07:01:01 zaius kernel: pinctrl core: initialized pinctrl subsystem
Jan 31 07:01:01 zaius kernel: usbcore: registered new interface driver usbfs
Jan 31 07:01:01 zaius kernel: usbcore: registered new interface driver hub
Jan 31 07:01:01 zaius kernel: usbcore: registered new device driver usb
Jan 31 07:01:02 zaius kernel: rtc-pcf8523 0-0068: rtc core: registered rtc-pcf8523 as rtc0
Jan 31 07:01:02 zaius kernel: usbcore: registered new interface driver usbhid
Jan 31 07:01:02 zaius kernel: usbhid: USB HID core driver
Jan 31 07:01:02 zaius systemd[1]: Listening on Process Core Dump Socket.
Jan 31 07:01:02 zaius systemd-sysctl[545]: Couldn't write 'fq_codel' to 'net/core/default_qdisc', ignoring: No such file or directory
Jan 31 07:01:15 zaius systemd[1]: Started Phosphor Dump core monitor..
Jan 31 07:01:15 zaius kernel[923]: [    0.140000] pinctrl core: initialized pinctrl subsystem
Jan 31 07:01:15 zaius kernel[923]: [    0.220000] usbcore: registered new interface driver usbfs
Jan 31 07:01:15 zaius kernel[923]: [    0.220000] usbcore: registered new interface driver hub
Jan 31 07:01:15 zaius kernel[923]: [    0.220000] usbcore: registered new device driver usb
Jan 31 07:01:16 zaius kernel[923]: [    4.120000] rtc-pcf8523 0-0068: rtc core: registered rtc-pcf8523 as rtc0
Jan 31 07:01:16 zaius kernel[923]: [    9.420000] usbcore: registered new interface driver usbhid
Jan 31 07:01:16 zaius kernel[923]: [    9.420000] usbhid: USB HID core driver
Jan 31 07:01:16 zaius kernel[923]: [   14.270000] systemd[1]: Listening on Process Core Dump Socket.
Jan 31 07:03:58 zaius systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Jan 31 07:03:58 zaius systemd[1]: Started Process Core Dump (PID 1338/UID 0).
Jan 31 07:03:58 zaius systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Jan 31 07:04:03 zaius systemd-coredump[1339]: Process 1034 (openpower-occ-c) of user 0 dumped core.
Jan 31 07:24:45 zaius systemd[1]: Started Process Core Dump (PID 1810/UID 0).
Jan 31 07:24:45 zaius systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Jan 31 07:24:50 zaius systemd-coredump[1811]: Process 1341 (openpower-occ-c) of user 0 dumped core.

root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.1-23"
VERSION_ID="v2.1-23-g0a8ea63"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.1-23"
BUILD_ID="v2.1"

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x02848000..0x02849000 !
[                                                  ] 0%open-power-zaius-v1.20-119-gb278ae9-dirty
        buildroot-2017.11-5-g65679be
        skiboot-v5.9-240-g081882690163
        hostboot-d18433a
        linux-4.14.13-openpower1-p2991974
        petitboot-v1.6.6-p2462045
        machine-xml-b0884b3
        occ-577915f
        hostboot-binaries-6924d6b
        capp-ucode-p9-dd2-v2
        sbe-095e608
[==================================================] 100%

```

PavamanSubramaniyam, 2018-01-31:
[20180131085617825473_Myffdc.zip](https://github.com/openbmc/openbmc/files/1681036/20180131085617825473_Myffdc.zip)

PavamanSubramaniyam, 2018-01-31:
[obmcdump_2_1517383490.tar.zip](https://github.com/openbmc/openbmc/files/1681117/obmcdump_2_1517383490.tar.zip)

mine260309, 2018-02-01:
Romulus has similar issue: https://github.com/openbmc/openbmc/issues/2745

gtmills, 2018-05-23:
#2745 is closed. @PavamanSubramaniyam @mine260309 Can this be closed?

gtmills, 2018-08-29:
Closing. Please reopen if still an issue. Thanks!

closed 2018-08-29 by gtmills