openbmc/openbmc #2857: Code update: Image priority setting post activation

gkeishin, 2018-01-31:
Looks like the user can’t know for sure when setting the priority value set is completed. 

It could take from few seconds to more depending on the scenario. So there is no real synchronous way to tell when it is done.

Post discussion with @anoo1 @geissonator we would need to provide something for the user using which it can query to know when done.

anoo1, 2018-02-06:
Setting the Priority value triggers a chain reaction to update the priorities for other versions and set the boot version via systemd service files that are async. The user can reboot the BMC before all the services are done, so ideally the setting the Priority value would monitor the service files and return until they're done.

mtritz, 2018-02-06:
There are journal entries that pop up when priority values are set - do we need additional indication beyond that?

amboar, 2018-02-06:
If this is something that takes time then you need to provide a progress indicator on the REST interface.

Having said that, how long does it take for the changes to complete? It seems surprising that it's long enough for people to trigger a race with a reboot.

anoo1, 2018-02-06:
It takes about 5 seconds, the issue has been seen during automation test cases, where the reboot is issued immediately after the call to set the Priority value returns.

> There are journal entries that pop up when priority values are set - do we need additional indication beyond that?
Yeah, we need to hold on returning because we can't have a requirement for the user to rely on the journal to determine if an operation has completed. Also to support automation.

mdmillerii, 2018-02-06:
5 seconds to change the priority is way to long.  It should be much less, we just need to erase and program 2 blocks to change the priority.

Is this including time to write the partition?

That being said, we still need to provide a way for rest to determine the activation has completed.

rfrandse, 2018-03-06:
<https://gerrit.openbmc-project.xyz/9418> Delay return from updateUbootEnvVars() until service file finishes
Resolves  #2857 Code update: Image priority setting post activation

anoo1, 2018-04-10:
This change has been reverted via https://gerrit.openbmc-project.xyz/#/c/10022/ because it causes intermittent issue https://github.com/openbmc/openbmc/issues/3069

closed 2019-04-27 by stale[bot]