openbmc/openbmc #2859: phosphor-networkd: changing NTPServers not taking effect

mine260309, 2018-02-01:
Changing [NTPServers](https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Network/EthernetInterface.interface.yaml#L33) in phosphor-networkd is expected to take effect immediately.

However, it just updates the config but not actually takes effect.

From the code, we can see that it tries to restart `systemd-timesyncd` service after the NTPServers are updated: https://github.com/openbmc/phosphor-networkd/blob/master/ethernet_interface.cpp#L509

Unfortunately, if `systemd-networkd` is not reloaded, `systemd-timesyncd` will not get the updated NTP server at all.

So it should actually reload `systemd-networkd` instead.

mine260309, 2018-02-01:
https://gerrit.openbmc-project.xyz/#/c/8767/

rfrandse, 2018-02-01:
<https://gerrit.openbmc-project.xyz/8767> Reload systemd-networkd on NTPServer changes
Resolves  #2859 phosphor-networkd: changing NTPServers not taking effect

closed 2018-02-06 by geissonator