openbmc/openbmc #2862: FFDC (dreport) data missing fw_printenv

bjwyman, 2018-02-01:
While attempting to patch a system for testing a change to the OpenBMC code, I noticed that the `/usr/local/share/pnor` mount was missing from the system.

Nothing immediately obvious as to why that was the case, so I ran the `dreport` tool to capture any information that should prove useful for later investigation. After rebooting the system, I was asked to look at the `fw_printenv` information. I expected that to be included in the `dreport` (FFDC) output, but alas, it was not.

Adriana (@anoo1) thought that capturing `fw_printenv` as part of `dreport` was probably a good idea. Usually, these settings should stick after a reboot, but these could be changed at some later time. This would be good data to include, especially for field support.

@ojayanth

ojayanth, 2018-02-02:
@bjwyman Good info to keep this as part of user imitated dump.

ojayanth, 2018-02-05:
Code is available for Review https://gerrit.openbmc-project.xyz/#/c/8827/

ojayanth, 2018-02-07:
Included Alt flash uboot environment details .
https://gerrit.openbmc-project.xyz/8879

closed 2018-02-08 by geissonator